smartavionics/Cura #161: 'Arange all models' packing distance

PhilBaz, 2022-03-08:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

With ABS it's nice to have some distance between parts. I currently mostly have to place them by hand.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

It would be nice to be able to set the packing distance with a setting. 10mm minimum distance apart, 20mm minim, etc. 

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Guess I could create false skirts around parts before exporting them, but that would make more work.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

People who use ABS that benefits from a little travel time for the tool head, and less reflected fan air from close by parts.

**Additional context**
(Add any other context or screenshots about the feature request here.)

smartavionics, 2022-03-09:
Hi, I've spent a few minutes looking at this and have drawn a blank so far. The code appears to set the spacing to a fixed 1.5mm but when I try and change that hard coded value to something larger, it fails saying that it can't fit all the objects in the build plate when it really should be able to. I will try and understand what's happening there.

PhilBaz, 2022-03-09:
Thanks for looking into it! I think it would be a useful feature. Its one of those things i'm slightly surprised isn't already in there.

smartavionics, 2022-03-09:
I now understand what was stopping a larger packing working and will think about how to allow the user to specify a value. At the same time, I am also going to look into being able to stop the packer from rotating parts.

smartavionics, 2022-03-09:
stopping a larger packing DISTANCE working

smartavionics, 2022-03-10:
I've made a new release that adds experimental settings to control the distance and also to disable the rotations. Changing those settings doesn't automatically trigger the rearrangement, you still have to do Ctrl-R or use the menu option. The arranging is still as quirky as before!

PhilBaz, 2022-03-10:
Oooo, love it!!!  Best part, disable rotation also works for model multiply... What a great addition. From my intal play around everything seems to be working perfectly. Thank you!

closed 2022-10-24 by PhilBaz