smartavionics/Cura #162: Rotate in 45 degree increments

d-portero, 2022-03-11:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
When rotating models, usually it is in increments of 45 degrees. It would be nice to have buttons or a text field where we could easily rotate the model instead of using the wheels. Many other slicers have this feature, but Cura has no similar option. 

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

smartavionics, 2022-03-11:
This (or similar) has been requested many times but the UM people won't do anything about it. It's not something I am interested in doing, sorry.

d-portero, 2022-03-11:
Just out of curiosity, is it particularly hard to implement?

smartavionics, 2022-03-12:
In the past when people have asked UM to add the ability to type the rotation angle in, the developers have come back saying there's some complexity involved. That may be bullshit because they can't be bothered or true, I have no idea.

See https://github.com/Ultimaker/Cura/issues/2779 for a discussion on this topic.

5axes, 2022-08-31:
Solution : Modify the RotateTool Plugin to get 45° Rotation and not 90° by Clic.
`angle = math.radians(90 if (self._active_widget.value - ToolHandle.AllAxis) % 2 else -90)`

`angle = math.radians(45 if (self._active_widget.value - ToolHandle.AllAxis) % 2 else -45)`