smartavionics/Cura #163: Bulging wall above large skin surface - Slicing?

PhilBaz, 2022-03-28:
**Application version**
(The version of the application this issue occurs with.)

Cura-mb-master-win64-20220310  ...but no suspicion its version specific.

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Win 10, AMD workstation with 1070ti. letting modesty slip, I would describe the PC as a total beast.

**Printer**
(Which printer was selected in Cura?)

Voron 2.4, tuned and printing beautifully.

**Reproduction steps**
1. (Something you did.)
Im working with a part that has large ( 8"x3" ) flat surfaces, followed by structures that rise above those surfaces. 

3. (Something you did next.)
Ive done a good bit of testing, looking closely at Gcode speed & flow. Fiddling with a few things. Nothing has solved it. Ill continue testing tonight. 1) Ill try printing with no difference in speeds between infill, inner, & outer wall, to see if it makes a difference. 2) I also plan to test pausing between the last skin layer and the next layer. Im curious if this is some sort of back pressure build up from the large skin surface. But dont know why it would be...

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)

No real steps here....just 1) slice, 2)slice looks good, no indication of weird slicing in the area 3) reproducible print artifact. 

Also, Its not just this one part. I'm printing others with different shapes, but similar large flat surfaces and gett a similar result.

**Expected results**
(What should happen after the above steps have been followed.)

Perfect print?

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

Ill post a file shortly.... Just want to test one or two more things....

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

Don't think its relevant.

**Additional information**
(Extra information relevant to the issue.)

Thank you!!!

PhilBaz, 2022-03-29:
Here is a file with the part and settings in Cura.... Tried some more testing but nothing changed. It almost seems to be coded in. hummm

[bulge above skin.zip](https://github.com/smartavionics/Cura/files/8367937/bulge.above.skin.zip)

smartavionics, 2022-03-29:
The only thing that occurs to me is that due to the large difference in print speeds you are getting over-extrusion for the lower speeds, i.e. when you print the top skin layer it's coming out thicker than it should. Does your printer support non-linear extrusion to compensate for the reduction in flow that occurs at higher print speeds? If it doesn't and you set up the extruder steps to be correct at a high print speed, when you print at a lower print speed, it will over-extrude.

smartavionics, 2022-03-29:
Oops, closed by mistake!

PhilBaz, 2022-03-29:
> 

I was thinking this might be the case as well, if I understand correctly. But I did some testing, pausing between the skin layer and next wall layer above. I paused, let the nozzle pressure normalize, extrude some filament, clean the nozzle, resume. The result was the same.

I think the test excludes the possibility of back pressure, speed changes etc. Do you think?   After my little pause clean routine it moves the tool-head into place and proceeded to extrude the same bulge. That's why I say it seems to be coded in.

Ill look into non-linear extrusion. I'm on klipper so not sure. I'm going to try another slicer (downloading super slicer) just for a control test.

Very curious....

smartavionics, 2022-03-29:
What I'm saying is that if the extrusion is correct when you print very fast, it will over-extrude when you print slow (like when you print the top skin) unless your printer supports non-linear extrusion. One of my printers had a Duet controller and that did support on-linear extrusion but I don't think it's a common feature in controllers.

shyblower, 2022-03-29:
@smartavionics I believe what you are referring to is called "Linear Advance" in Marlin firmware and "Pressure Advance" in Klipper.

PhilBaz, 2022-03-29:
@smartavionics 

But in my testing didn't I eliminate that dynamic by doing the following? (maybe not)

"pausing between the skin layer and next wall layer above. I paused, move the nozzle away, let the nozzle pressure normalize, extrude some filament, clean the nozzle, resume. The result was the same."

By doing the above I feel like I've eliminated any dynamics that roll from one phase (skin) into the next (wall above). If the nozzle returns to the Code an extrudes the bulge.

Yes, if its pressure advance, then I have the tuned and active.

smartavionics, 2022-03-29:
No, I'm not talking about linear advance or pressure advance, I'm talking about non-linear extrusion as supported by the M592 gcode on the Duet...

https://docs.duet3d.com/User_manual/Reference/Gcodes#m592-configure-nonlinear-extrusion

What I'm thinking is that the top skin is over-extruded which makes it thicker so when the next layer is printed, the distance between the nozzle and the layer below is too small so the walls bulge?

shyblower, 2022-03-29:
@smartavionics seems to tackle the same issue, though. Linear Advance also modifies the amount of extrusion, depending on printing speed but uses only one parameter, K, which expresses the of filament compression length per velocity unit in mm/mm/s.
... but I'm neither a native English speaker not a hardware engineer, so I might misinterpret its documentation.

smartavionics, 2022-03-29:
Linear advance, if I understand it correctly, is all about improving the change in extrusion rate that occurs when the print speed changes. i.e. when a sharp corner is approached, the extrusion rate is reduced to lower the nozzle pressure to avoid over-extrusion as the nozzle speed slows down. Similarly, the extrusion rate is raised while the nozzle is accelerating away from the corner.

But non-linear extrusion is not that. It simply modifies the extrusion rate in a non-linear way with respect to the speed. For example, if non-linear extrusion is not being used and for a print speed of, say 50mm/S the extruder is turned at, say 80 steps/S, and then if the speed is doubled to 100 mm/S, the extruder rate will be doubled to 160 steps/S.

When the non-linear extrusion feature is used, the extruder rate is increased non-linearly as the print speed is increased so, using the above example numbers, with non-linear extrusion active, the extruder step rate when printing at 100 mm/S could be 180 rather than 160 and that compensates for the flow reduction that otherwise occurs as the print speed is increased.

Of course, you can't keep increasing the extruder rate as it will just start skipping or jamming but it is possible to widen the print speed range over which the extruder will operate reasonably linearly.

So back to my original thought as to what is causing the problem you are reporting, I find it hard to believe that your extruder will be printing equal amounts of filament per mm of nozzle travel when you are printing walls at 160 mm/S and the top skin surface at 45 mm/S. And if you are not seeing under-extrusion on the walls, it surely must be over-extruding the skin?

PhilBaz, 2022-03-29:
@smartavionics  

Ahhh, now Im clear. You think the skin is over extruding at a slow speed leading to increased thickness of the skin, and resulting in squashing (bulging) in the following layer. Im just getting the point of what you were saying. That would be consistent with my testing result. Okay, now I have new tests to run. 

Ill report back. 

PS....I understand the need for non-linear extrusion. With my foray into highspeed/ flow printing I can confirm that speed and flow non-linear. Higher speed prints need higher flow rates to achieve requested line width. ill have to look into whether Klipper has this integrated and how to incorporate it into my slicing.  

Thanks for staying with my issue !!! Ill let you know when/how I get it sorted.

PhilBaz, 2022-03-30:
@smartavionics 

I just realized something that I think negates the working theory of fat skin....

The bulging wall is not actually above skin. I have been reducing top & bottom skin expand distance to zero. Started doing that half way through my initial testing. So the bulging wall is just stacking on top of more wall. 

I also did another round of testing... inner, outer wall & skin all set to the same speed and flow (120mm @ 89% flow, my calibrated flow rate), then same with skin dropped to 84%. It wasn't till after the 84% I realized the wall wasn't sitting on skin... ill attach images of the two relevant layers.

I'm keen to chase this bulge down the rabbit hole if you have other ideas? Ill also run some tests with super slice tomorrow as a reference.

I'm running a test right now with zero top skin layer to see what happens....curious.

[image]
[image]

shyblower, 2022-03-30:
@PhilBaz 
> Ahhh, now Im clear. You think the skin is over extruding at a slow speed leading to increased thickness of the skin, ...
But this is exactly what "Pressure Advance (Klippper)" and "Linear Advance (Marlin)" are intended to solve!
These algorithms non-linearly increase flow when the speed increases, where the K paramter is used to specify the filament compression length per velocity unit, in mm/mm/s.

PhilBaz, 2022-03-31:
@smartavionics 

Okay, some interesting results.... Ill lay out the tests below. I'm slightly perplexed. Curious to hear your thoughts. 

1) Test: No top skin - Result: perfect wall

2) test:  5 'extra skin walls' & 'skin removal width' adjusted until the skin walls are just touching the external wall stack on the above layer - Result: bulge above skin layer.

3***) Test: 'skin removal width' adjusted until the skin is like 10-15mm away from the wall stack above Result: bulge above skin layer.

Test 3 leaves me completely perplexed. In effect its essentially the same as test 1 as far as the walls staked above the skin layer are concerned.  Id be curious to hear your thoughts if you have a minute.

PS..I drew some lines on the image so as not to confuse the bulge with the brim remnants.

[image]

[image]

smartavionics, 2022-03-31:
That is interesting. I don't have any ideas at the moment but will continue to ponder what's happening here...

PhilBaz, 2022-03-31:
@smartavionics 

Yeah, funny right. 

I've been printing these in ABS, as that's the material for the end use. But ill double check the 1&3 results with PLA today. Want to make sure its not a weird ABS warping issue what would be hard to understand/anticipate. I don't think it is, but I'm out of ideas at this point.

qwerty8224, 2022-04-02:
I have been observing this behavior for a long time and it is not related to the speed of printing .... but I can’t understand what’s the matter ...

PhilBaz, 2022-05-20:
@smartavionics 

Just had my first productive thought on this in a while... I'm going to manually combine the G-code from the problematic slice (wall bulging above skin layers) and the slice without skin that does not exhibit the bulge and see what happens...

I think I already ruled out nozzle pressure & material dynamics....either way this should indicate whether the problem (bulge) is baked into the g-code......I think?