smartavionics/Cura #164: "top surface skin speed" and "infill speed" are ignored when infill density is set to 100%

shyblower, 2022-04-05:
**Application version**
20220310 and all earlier versions I tested

**Platform**
Linux x86_64

**Printer**
Original Prusa MK3

**Reproduction steps**
2. Set "top surface skin layers" to at least 1.
3. set "top surface skin speed" and "infill speed" different to "top/bottom speed".

**Screenshot(s)**
Layer 28:
[image]

Layer 29:
[image]

**Actual results**
All skin layers, except the "initial layers", are printed with "top/bottom skin" settings,  and "top surface skin" and "infill" specific settings are ignored.

**Expected results**
"top surface skin layer" and "infill" specific configuration should be honored, even with an infill density of 100%.

**Additional information**
This works as expected when setting the infill density to 99,999% (zigzag) which is my quick&dirty workaround.

smartavionics, 2022-04-06:
They decided that 100% infill should be done as skin so they force the number of top layers to 0 and the number of bottom layers to 999999. What a hack!

You can get the top layers back by simply overriding the number of top layers. So where it shows 0 top layers, put the value that was previously displayed when the infill density was < 100%.

There's no better workaround to get the 100% infill skins printed using the infill speed than doing what you did by setting the density to slightly less than 100%.

Personally, I never use 100% infill, the highest I go is 90% because I can't be 100%! sure that my extruder calibration is accurate enough to guarantee that I won't get an over-filled object.

shyblower, 2022-04-06:
Thank you for the explanation!
Your workaround also sounds much better than my 99,999% solution.

closed 2022-04-06 by shyblower