smartavionics/Cura #165: Cura x64 Won't Launch on x64 Win10 or Win11 without newest Microsoft Visual C++ Redistributable

TechNuggets, 2022-04-17:
**Application version**
4.13.1 Basically all of the Windows x64 versions since 2021

**Platform**
Windows 10 and Windows 11

**Printer**
N/A

**Reproduction steps**
1. Installed Cura, with C++ redis checked.
2. Cura launches with an error

**Screenshot(s)**
Gives an error saying it is unable to find a DLL and describes a file under C:\Users\Brunto\etc.etc.etc.

**Actual results**
Doesn't Launch Cura

**Expected results**
Cura should launch

**Project file**
N/A

**Log file**
N/A

**Additional information**
Installing the newest VC package from here https://docs.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170  fixes the issue on both Windows 10 and Windows 11.

I believe the included redistributable package in the installer is outdated and should be updated to the above.

smartavionics, 2022-04-17:
Understood. I'll look into this when I next make a release.

smartavionics, 2022-05-07:
Hi, the latest release now contains the more recent vcredist file.