smartavionics/Cura #170: Settings guide plugin often crashes Cura.

PhilBaz, 2022-05-05:
**Application version**
(The version of the application this issue occurs with.)

current

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10

**Printer**
(Which printer was selected in Cura?)
Voron 2.4
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Right click on parameter, select settings guide in flyout. Settings guide full version appears, but Cura locks up after settings guide is closed.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

Lock up

**Expected results**
(What should happen after the above steps have been followed.)
smooth operation...

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
settings guide plugin may be the culprit. 

PS...saw you chime in on my Cura 5 stuttering issue...hope they integrate your solution.

smartavionics, 2022-05-05:
You need to raise the issue with the author of the settings guide.

PS - they could have fixed their version of simplify() years ago but have chosen not to.

PhilBaz, 2022-05-05:
Ill do that...

Thanks again smart!

PhilBaz, 2022-10-24:
I'm not sure if this ever got fixed but I saw the problem was located on the plugin side. 

https://github.com/Ghostkeeper/SettingsGuide/issues/80