smartavionics/Cura #171: Rounding Error on perimeter walls

McNugget6750, 2022-05-08:
**Application version**
This fork does not provide a build number...

**Platform**
Windows 10

**Printer**
Malyan M180 custom build 

**Reproduction steps**
1. Open a file that has a volume with a cutout and print the top surface of the volume at an angle
2. Notice that the inner perimeter, when the angled surface of the volume breaks up a full circumference and the head needs to turn around instead of continuing counter clock wise, will be printed one perimeter width inward. This is even visible in the slicer preview already.

**Screenshot(s)**
[image]

**Actual results**
[image]
[image]
[image]
[image]

**Expected results**
Smooth surface from the rim of the model all the way to the mounting surface of the front plate.

**Project file**
[20220424 - Speakers.zip](https://github.com/smartavionics/Cura/files/8647211/20220424.-.Speakers.zip)

**Log file**
[cura_log.zip](https://github.com/smartavionics/Cura/files/8647215/cura_log.zip)

**Additional information**
It would be nice if we could figure this one out as I have seen this many times on other prints, too, and I always needed to fix this in CAD by adding some sort of 0.1mm-0.2mm offset to those surface. This is the first time I actively looked for it in the slicer and it's clearly visible that the perimeter moves one width inwards for no apparent reason.

smartavionics, 2022-05-08:
Hi @McNugget6750 , what's happening here is that you have specified a non-zero value for hole horizontal expansion and so as the big cavity in your model is a hole, it gets expanded. You can stop this happening by putting in a value for Hole Horizontal Expansion Max Length which will stop holes whose perimeter length is >= that value from being expanded. Hope this helps!

[image]

McNugget6750, 2022-05-08:
Wow, that does make a lot of sense! Thank you for the quick reply and excellent thinking! I'll try that!

McNugget6750, 2022-05-08:
Yes, that fixed it! Thanks a lot!!

closed 2022-05-08 by McNugget6750