smartavionics/Cura #174: Bridging gets its own skin overlap setting

PhilBaz, 2022-05-21:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Yes, in a sense.... bridging might benefit from additional skin overlap, for anchoring. I'm specifically thinking of internal bridging. I use bridge settings to establish a first skin layer.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Skin overlap setting for bridging.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

using normal skin overlap setting, but then you change the function and aesthetic of non-bridge skin areas as well. 

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

all?

**Additional context**
(Add any other context or screenshots about the feature request here.)

smartavionics, 2022-05-21:
Do you have an example project that illustrates how the current bridge skin overlap is inadequate?

PhilBaz, 2022-05-22:
Well its not that its inadequate....i can pump it up. Just not without affecting the final surface quality on the non bridge skin. Just more that I think there is a strong rational for decoupling it from the normal skin overlap setting. 

I have been having issues getting it to stick to the initial perimeter.  But then I'm printing in ABS and trying not to use much fan (warpy ABS), that might otherwise help.  I'm thinking increased overlap might give a better chance of holding the bridge from that initial edge.  

fighting with this part...funny enough the one I have on hand is having more problems near the center because I have a already bumped up the skin overlap.  I wonder if skin overlap of even 200% might be good for bridge. That much overlap causes real problems with expanded skin regions I have come to find.

[image]

smartavionics, 2022-05-22:
Can you please provide the project file?

smartavionics, 2022-05-22:
FYI, I have done some work on this and my next release will have a bridge skin overlap setting.

PhilBaz, 2022-05-24:
@smartavionics Awesome! Thats great.... sorry for the radio silence. Its a long weekend here, was planning to send something tomorrow...

PS...dont know if you saw the response on that stuttering ticket. https://github.com/Ultimaker/Cura/issues/12090#issuecomment-1134408790 . Ill send them something tomorrow. Seems doubtful it got worked out between beta and release, but who knows.

PhilBaz, 2022-05-26:
Thanks for chiming in on corner preference....!

PhilBaz, 2022-05-26:
PS....The flow rates in bridging those are absoluter flow rates for bridge wall and bridge skin, or they further modify the flow rates for walls and skin?

smartavionics, 2022-06-18:
See https://github.com/smartavionics/Cura/releases/tag/4.20.1

closed 2022-10-24 by PhilBaz