smartavionics/Cura #175: Feature Request: Build appimages on bionic

theofficialgman, 2022-05-31:
**Is your feature request related to a problem? Please describe.**
Unable to run on Ubuntu bionic (eg: nvidia jetson devices)

**Describe the solution you'd like**
Build the arm64 (and armhf) appimages on ubuntu bionic to avoid GLIBC 2.28 being necessary from building on buster.

smartavionics, 2022-05-31:
Hi, is it straightforward to install glibc 2.28 on bionic? Does doing that cause any other issues? If not, then it would be better to stay as we are now so that people who are still running buster can use the appimages.

theofficialgman, 2022-05-31:
> Hi, is it straightforward to install glibc 2.28 on bionic? Does doing that cause any other issues? If not, then it would be better to stay as we are now so that people who are still running buster can use the appimages.

No, attempting to install glibc 2.28 will break almost many other packages on bionic. That's basically the most damaging thing you can do to the os. Glibc is the base c library that every single package in the os depends on.

Building on bionic shouldn't cause any issues with using the appimage on buster (or bullseye which many people do).

theofficialgman, 2022-05-31:
There are 18.04 server images here for the pi
https://wiki.ubuntu.com/ARM/RaspberryPi
You can install the desktop on that with accelerated graphics following the couple step instructions if you would like

smartavionics, 2022-05-31:
Hmm, the 18.04 images are not for the pi 4 so I can't use them.

theofficialgman, 2022-05-31:
@smartavionics sorry its not directly on that page. thats just documentation.
you can find the 18.04 rpi4 image here http://cdimage.ubuntu.com/ubuntu/releases/bionic/release/

smartavionics, 2022-05-31:
OK, thanks. I'll look at doing that when I can. Quite busy at the moment so I can't say when it will get done.