smartavionics/Cura #176: ununtu 18 missing libraries for cura version May 7, 2022 version of cura

gr5, 2022-06-09:
**Application version**
Cura-mb-master-x86_64-20220507

**Platform**
ubuntu 18.04.6 LTS

You mentioned that there may be some libraries missing for earlier versions of ubuntu.  I'm actually going to upgrade now to ubuntu 20 so here is what I got in ubuntu 18 just now.

geo@geo-Wild-Dog-Pro:~/Downloads$ ./Cura-mb-master-x86_64-20220507.AppImage 
Traceback (most recent call last):
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
  File "/home/markb/3dp/cura/cura-build.git/build/inst/bin/cura", line 28, in <module>
ImportError: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/lib/Arcus.so)
Cura crashed, was that because a required library couldn't be found?
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.28' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libQt5Gui.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.28' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libpng16.so.16)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libsystemd.so.0)
/tmp/.mount_Cura-ml8awQG/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.30' not found (required by /tmp/.mount_Cura-ml8awQG/usr/bin/libsystemd.so.0)
	libxcb-xinput.so.0 => not found

eGit, 2024-01-06:
Same problem. GLIBC missing.

The current official Cura version 5.6.0 works for me.

`./Cura-4.20.16-mb-x86_64.AppImage 
Traceback (most recent call last):
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
  File "/home/markb/3dp/cura/cura-build.git/build/inst/bin/cura", line 30, in <module>
ImportError: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/lib/../libpython3.8.so.1.0)
Cura crashed, was that because a required library couldn't be found?
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libQt5Gui.so.5)
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libQt5Core.so.5)
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libpng16.so.16)
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libsystemd.so.0)
/tmp/.mount_Cura-4w4rAyC/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.30' not found (required by /tmp/.mount_Cura-4w4rAyC/usr/bin/libsystemd.so.0)
	libxcb-xinput.so.0 => not found`

smartavionics, 2024-01-06:
Hi @eGit , what OS are you using?

eGit, 2024-01-06:
Linux Mint 19.3 is Ubuntu 18.04.6 LTS under the hood. 

Distributor ID:	LinuxMint
Description:	Linux Mint 19.3 Tricia
Release:	19.3
Codename:	tricia

eGit, 2024-01-06:
The official Cura shortly changed their build system, so version 4.13 to 5.x was not compatible due to GLIBC problems.

But starting around 5.5 (maybe 5.4 or 5.3) they fixed the problem. It all depends on the build system. Their is no reason to not support the old GLIBC version.

smartavionics, 2024-01-06:
Thanks for the info. Sorry, I'm not changing my build system. Maybe there's a workaround whereby you put a more recent glibc in a directory and use LD_LIBRARY_PATH to force the appimage to use that rather than the system glibc?

eGit, 2024-01-06:
I doubt it. This topic was broadly discussed in the Cura community (forum and Github) for a while and nobody of the devs suggested trying it.

The people at Cura did get a new build server which than did break Cura for 18.04 and earlier. Around Cura 5.5 they fixed it, because too many people were still on GLIBC 2.28

smartavionics, 2024-01-06:
Oh, it did work on 18.04 as that was what I was using for some years before I moved to 20.04.

eGit, 2024-01-06:
Thanks. Do you know which is the last version that worked?

Great project btw. Will you ever switch to Cura5? Or will you wait for Cura6?

smartavionics, 2024-01-06:
The last release before I moved to using 20.04 was https://github.com/smartavionics/Cura/releases/tag/20210629.

I very much doubt that I will merge my mods into a later upstream Cura version. I'm just maintaining what I have already done and will keep fixing bugs that I become aware of. The only Windows computer I have is a creeky laptop and when that dies I won't be able to build Windows releases any more.

eGit, 2024-01-07:
Thanks!

OffTopic: is there a mechanism to "hack" into Cura to achieve what one wants? I know there is the Plugin-Mechanism. Maybe one could write a plugin for Cura to implement quick hacks or easily transfer some of the functionality in your fork back to Cura, without a real merge?

For example for the first layer I would like to print the outer walls of ALL objects first before inner walls and the bottom fill. Also I want the outer walls to be printed much slower, inner walls and filling normal speed. For me, if the outer walls of all objects stick perfectly, the print will be a success.
Having many bigger objects is a problem, because if prints the entire first layer of the first object. Then the second object and so on. Since I print the first layer very slow, this takes some time. If ONE object fails, I have to stop the print.
FAIL FAST is a programming paradigm. That's why I want to print just ALL the outer walls of all objects first and then the rest.

gr5, 2024-01-07:
A quicker way to summarize what you want:  "all at once" mode for the first layer then "one at a time" mode for the rest.

eGit, 2024-01-07:
"all at once" mode for the first layer then "one at a time" mode for the rest.

We are talking about your fork or the official Cura?

gr5, 2024-01-07:
Don't get me confused with "smartavionics".  You and I are writing on the github issue area for the smartavionics fork of Cura.  You should also post this suggestion on the ultimaker/cura issues area!  You never know who is going to like your idea enough to write the code for it.

Lately smartavionics (as he says above) hasn't been adding any features.  So you might have better luck with the UltiMkaer Cura team.  It is very little effort for you to go over there and request the same thing:
"all at once" for the first layer and then "one at a time" mode for the rest of the print.  Explain why it's a useful feature.  I agree that it is a useful feature.  You can then post a link here to that post and I can chime in saying something like "this sounds like a great feature".

eGit, 2024-01-07:
Thanks. You are right. I got you confused.
I think I saw similar requests/issues yesterday when I was looking for more control over the first layer. I think to remember, that the Cura devs noticed that there is a desire for fine-grained first layer control (which smartavionics-
Cura does to some extend). But they said that this change won't be easy and that they are already duplicating a lot of stuff for the first layer and that the way it is all done is not really nice at the moment.