smartavionics/Cura #180: Cura-4.20.1-mb-armhf.AppImage doesn't carry over the settings from Cura-mb-master-armhf-20211110.AppImage

Jens5395, 2022-07-13:
**Application version**
Cura-4.20.1-mb-armhf.AppImage

**Platform**
Pi4 

**Printer**
(Which printer was selected in Cura?)
Not applicable (any/all)

**Reproduction steps**
run the appimage

**Actual results**
(What happens after the above steps have been followed.)
Cura starts out as a brand new installation

**Expected results**
(What should happen after the above steps have been followed.)
Previous upgrades always used the setup from the old version

mard89, 2022-07-31:
I would like to note that the appimage builds do not seem to create the usual config folders; I'm running the 4.20.1 64bit appimage and no 4.20 folders have been created under ~/.local/share/cura or ~/.config/cura

smartavionics, 2022-07-31:
I just tried using the x86_64 appimage and it created both of the required 4.20 directories. What system are you using?

smartavionics, 2022-07-31:
And I just tried the aarch64 image and that created those directories also.

mard89, 2022-08-01:
Just double checked.
On an ubuntu 21.10 install no 4.20 folder was created and also no master folder from previous versions.
On an ubuntu 22.04 install all seems to be working as expected.

smartavionics, 2022-08-01:
Does your environment contain XDG_CONFIG_HOME and XDG_DATA_HOME? Those are used instead of ~/.config and ~/.local/share if they are defined.

mard89, 2022-08-01:
Not that I know of. There is nothing weird in /etc/environment or /etc/profile.d/*

Let me check a fresh VM to make sure it's ubuntu 21.10 and not something wrong with this specific install

smartavionics, 2022-08-01:
Try "set | grep XDG_" and see what gets printed...

mard89, 2022-08-01:
Neither is set. But they are also not set on the 22.04 install.

I just realized that this whole discussion is moot. Ubuntu 21.10 went end of life at the 14th of July, so i will need to update it anyway. Hopefully restoring the behaviour as seen on 22.04

smartavionics, 2022-08-01:
Hmm, weird, though. Would be nice to understand what's happening there.

mard89, 2022-08-02:
I figured it out. As expected, the bug was me and not the appimage.

Previously I have been trying different ways to get appimages to integrate into better into my system. Nothing really worked, I gave up and didn't try any of the plug-ins or packages on the 22.04 install.
While searching for the config folders to back them up before wiping the 21.10 install I found the following folders:

```
martijn@martijn-ThinkPad-T440:/$ find ~ -type d -name "cura" -print
/home/martijn/.config/cura
/home/martijn/.cache/Ultimaker B.V./cura
/home/martijn/.local/bin/appimage-firstrun/.config/cura
/home/martijn/.local/bin/appimage-firstrun/.cache/cura
/home/martijn/.local/bin/appimage-firstrun/.cache/Ultimaker B.V./cura
/home/martijn/.local/bin/appimage-firstrun/.local/share/cura
/home/martijn/.local/share/cura
```

appimage-firstrun was one of the methods I tried. Not sure why the folders are placed where they are now considering the fact that all this was before I switched over to this fork of cura; I've always opened "cura master" as a stand-alone file.

smartavionics, 2022-08-02:
Thanks for the update. Mystery (mostly) solved!