smartavionics/Cura #181: Multiple layers for Initial Expansion

5axes, 2022-07-22:
There is a feature in SuperSlicer that I love and use every time, It's the Initial expansion on Multi-Layer. This option is not present on Cura ant it would be a very nice feature.

This feature is adding a new parameter : Nb of layer for the initial expansion

The number of layers on witch the initial expansion will be active. The first layer will be shrunk by the Initial **Layer Horizontal Expansion** , then the next layer will be gradualy change up to the layer indicated by this parameter to reach the normal **Horizontal expansion.**

[image]

smartavionics, 2022-07-22:
Sounds easy enough, I'll take a look...

smartavionics, 2022-07-25:
Will be in the next release...
[image]

smartavionics, 2022-08-05:
Now in https://github.com/smartavionics/Cura/releases/tag/4.20.3

5axes, 2022-08-05:
Thanks a lot !

closed 2022-08-05 by 5axes