smartavionics/Cura #182: Fuzzy Skin (noise) for vase mode

PhilBaz, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

In a sense. Fuzzy skin doesn't work with vase mode.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Some way to add fuzzy skin like noise (x-y coordinate variation noise) to vase mode prints. To create a finely textured surface.... ***When using vase mode.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

I have achieved this by using mesh displacement in my poly-modeling phase to achieve this fine texture.  This requires super high-poly models to achieve the desired fine grain surface texture texture.

In non-vase models I can use the fuzzy skin feature to crate a fine surface noise that is perfect.

Perhaps this would be a giant pain in the butt. or maybe its easily implemented? I wouldn't know. It would be a sweet feature though. Fine surface noise can really transform the look of a print and the visual perception of the material. It can also hide flaws that might otherwise be glaring eye-sores.

smartavionics, 2022-08-17:
Yeah, that sounds interesting, I'll put some thought into it and report back....

smartavionics, 2022-08-17:
You can combine fuzzy skin and vase mode (in my builds, don't know about Cura 5). Does that not provide what you need?

[image]

PhilBaz, 2022-08-17:
Ahhh.... I must have tried in Cura 5. I'm going back and forth these days. Sorry, yes, that is all I need then. 

Thank you!

Still have my fingers crossed for Cura 5 master branch!

smartavionics, 2022-08-18:
Yeah, they seem to have completely wrecked the vase mode in 5, only allows concentric pattern bottom layers, object must have holes removed, no fuzzy skin.

PhilBaz, 2022-08-18:
@smartavionics 

I know. I think I was the first to report the concentric issue. Think the first few responses said I was mistaken. Also, seams are a mess. Messy seams have a surprisingly large impact on print quality.

closed 2022-10-24 by PhilBaz