smartavionics/Cura #183: 4.20.4 Visible Top Skin Has Only One Contour

5axes, 2022-08-31:
Hello,

First of all, thanks again for the work done to integrate this new options. My post is not really about an issue but more a remark about the result compared to SuperSlicer. Not having tested both on a machine yet, I can't say which is the better solution. So it's more an open discussion than a real request.

The parameter works well, but in the interface layer the contour is offset to the outside, whereas in SuperSlicer it is created towards the inside. This should give a better overlap of the infill. 

**Cura 4.20.4**

[image]
[image]

**SuperSlicer 2.5.59.0**

[image]
[image]

[OneContour.zip](https://github.com/smartavionics/Cura/files/9462139/OneContour.zip)

smartavionics, 2022-08-31:
Thanks for the prompt feedback. Yes, what happens where the skin meets those inner walls and the infill is still not handled optimally. More work is required...

smartavionics, 2022-08-31:
I've probably fixed that but will test some more before making a release. I may well just update the 4.20.4 binaries rather than make a new 4.20.5 release but I'll let you know when it's done...

[image]
[image]

smartavionics, 2022-08-31:
I've just updated the 4.20.4 binaries. There's still an issue with unwanted skin being generated when the skin expand distance is greater than the total width of the inner walls but I haven't been able to fix that yet.

smartavionics, 2022-09-01:
I've fixed all the known issues with this new feature and updated the 4.20.4 builds. Please download again.