smartavionics/Cura #184: 4.20.4 Issue with monotonic order and ZigZag Pattern

5axes, 2022-09-01:
I have an issue with your release of Cura , it's not a particular problem related to the last release, but I never notice thiw point before.

Conditions : 
Monotonic Top/Bottom Order : On
Top/Bottom Pattern : ZigZag.

In this case  the pattern generate a lot of movement from one side to the other of the layer.
[image]

On the standard release of Cura the Monotonic order is correct.
[image]

Sample Case 
[OneContour.zip](https://github.com/smartavionics/Cura/files/9473316/OneContour.zip)

smartavionics, 2022-09-01:
When you say "standard release" which version are you referring to? Remember, my builds are based on 4.13 and not 5.

5axes, 2022-09-01:
Yes I'm Sorry. You're Right the 4.13 & 4.12 release of Ultimaker have the same Issue.

Solved in 5.1  ( Still a mess in 5.0)

smartavionics, 2022-09-01:
No problem, I'll see if I can work out what they did to fix it and, if possible, will incorporate the fix.

smartavionics, 2022-10-04:
Looks like this problem hasn't really been resolved yet https://github.com/Ultimaker/Cura/issues/13456

5axes, 2022-10-04:
Not sure , in this issue we have a problem with the ZigZag pattern. In the mentionned issue we have a linear pattern. Could be a different problem.