smartavionics/Cura #186: 'Random' seam isn't at all random

nirurin, 2022-09-18:
**Application version**
Latest (4.20.4)

**Platform**
Windows 11 

**Printer**
Ender5+

**Reproduction steps**
1. Add a shape
2. Set random seam
3. See that the seams are not random

**Screenshot(s)**
[image]

**Actual results**
Mostly linear vertical seams

**Expected results**
Randomly scattered seams

**Additional information**
Take a look at prusaslicer 'random' or superslicer 'scattered' for what a random seam should be.

smartavionics, 2022-09-18:
I never use the random mode but I tried a quick test and it looks random, could you please provide a project file (do File->Save and then rename the .3mf file to .zip or zip it up and attach to this issue). Thanks.

[image]

smartavionics, 2022-09-18:
I think that different systems (Windows, Linux, etc.) have different random number generators. I've taken a look at how Cura generates and uses random numbers and have made some changes which are going to be in my next release (4.20.5) so perhaps you can try that and let me know if you get better results.

nirurin, 2022-09-18:
> I think that different systems (Windows, Linux, etc.) have different random number generators. I've taken a look at how Cura generates and uses random numbers and have made some changes which are [...]

As requested, here is a project file of an example, along with a picture of the result I get. As you see in the picture, it comes with a very linear seam (basically 75% of the model is one straight seam). 
I can't imagine this is anything to do with a random number generator... even if windows is that badly written, it wouldn't come up with the exact same number twenty times in a row surely?! Though that would explain why both this and cura5 seem to act the same way. I guess prusa/superslicer use a different method for randomness. 

Happy to try the new version though, any idea when it might be available for testing? 

[CE5P_cross-blade-sansCylinder_001.zip](https://github.com/smartavionics/Cura/files/9594932/CE5P_cross-blade-sansCylinder_001.zip)

nirurin, 2022-09-18:
New version seems to work, it's not as 'scattered' as the superslicer offers (I assume it uses additional logic to prevent the 'random' seam from ending up near previous layers), but it's at least as good as prusaslicer offers, and is much better than cura5! And is more than sufficient for my needs, thankyou!

Unrelated to the topic (as this can be closed now) but I'm curious to ask - are there any plans for arachne on this build? I'm still undecided on the benefits over the old engine, but I was just curious if you thought it was improved enough to warrant merging?

smartavionics, 2022-09-18:
> So I assume even the 'random' seam logic is still forcing seams onto edges?

The random seam mode doesn't introduce extra points to the polygons so if you are slicing a rectangle then there's only 4 points to choose from (the corners!)

> Unrelated to the topic but I'm curious to ask - are there any plans for arachne on this build? I'm still undecided on the benefits over the old engine, but I was just curious if you thought it was improved enough to warrant merging?

Initially, I thought that I would merge Arachne after they had removed the worst of the bugs but I'm thinking now that it just isn't worth it. Also, because the build process has changed a lot for Cura 5, I probably wouldn't be able to continue to build for the Raspberry Pi and I would like to keep supporting that platform if possible.

nirurin, 2022-09-18:
> The random seam mode doesn't introduce extra points to the polygons so if you are slicing a rectangle then there's only 4 points to choose from (the corners!)
>
>
> Initially, I thought that I would [...]

Interesting, I guess this means that prusa/super must add more geometry, as their 'random' will put seams randomly around the whole model, including all the flat-sides of cubes. Interesting. 

I am still undecided on arachne, as I haven't been able to test it on detailed models yet, are there any settings you'd recommend in the 'classic' engine you use that would allow for arachne levels of clean detail?

smartavionics, 2022-09-18:
> Interesting, I guess this means that prusa/super must add more geometry, as their 'random' will put seams randomly around the whole model, including all the flat-sides of cubes. Interesting.

I do a similar thing for my "exact" z-seam mode, I effectively add a point into the polygon where I want the seam to occur. In reality I don't add a point but the effect is the same. It would be very easy to extend that to provide a truly random z-seam. I'll look into it....

> I am still undecided on arachne, as I haven't been able to test it on detailed models yet, are there any settings you'd recommend in the 'classic' engine you use that would allow for arachne levels of clean detail?

Well, Cura 4 and below can't cope nicely with walls that are odd numbers of wall line widths so it always pays to either modify the width of walls in the model or tweak the line widths in Cura.

smartavionics, 2022-09-18:
Turns out, it was easy to do...
[image]

nirurin, 2022-09-18:
> I do a similar thing for my "exact" z-seam mode, I effectively add a point into the polygon where I want the seam to occur. In reality I don't add a point but the effect is the same. It would be very easy to extend that to provide a truly random z-seam. I'll look into it....
> 

I'd appreciate that. While on models with corners it's usually better to put the seam on a corner, on rounded models it's much better to have the seam scattered as it hides it in the 'noise' (especially if the seam is calibrated to be as small as possible). 

While I was typing this, you send the picture of you accomplishing it! Haha thankyou
 
> Well, Cura 4 and below can't cope nicely with walls that are odd numbers of wall line widths so it always pays to either modify the width of walls in the model or tweak the line widths in Cura.

I'll have to do some testing with and without arachne to see how much different it makes to the prints I do haha. Thankyou

smartavionics, 2022-09-18:
However, I've just discovered a regression in 4.20.5 that could cause crashes so I have removed the builds from github and will replace with fixed versions tomorrow.

nirurin, 2022-09-18:
Not a problem, I'll keep an eye out for them, thank you for your help!

So in your opinion arachne isn't of much benefit? I currently use 0.6mm nozzles, but it's hard to find decent test cases for it haha

smartavionics, 2022-09-19:
Hi, I've replaced the builds, shouldn't crash now. BTW, this build does distribute the random start points along the straight edges as my image above shows. I realised after building it that it is still not completely random because the probability of a start point being on a given edge is equal no matter how long the edges are, i.e. a short edge is just as likely to have a start point on it as a long edge. I've fixed that for my next build.

Compare these two images to see the difference in distribution of the start points...

This is the current 4.20.5 build:
[image]

This will be in the next build:
[image]

smartavionics, 2022-09-19:
> So in your opinion arachne isn't of much benefit? I currently use 0.6mm nozzles, but it's hard to find decent test cases for it haha

TBH, I've not used Cura 5 at all so I can't say it's of no benefit. Undoubtedly, there will be some models and configurations that would indeed benefit by using Arachne.