smartavionics/Cura #187: Missing library

Jens5395, 2022-09-25:
I have had issues running the latest Cura versions for some time now. Today I decided to try and figure out why Cura 4.13 was running fine but 4.20.5 did not start. This is on a x86 based laptop running Ubuntu 22.04.
As it turns out, the appimage seems to be miissing  the libxcb-xinput-dev package. Once installed, 4.20.5 started and ran just fine.

smartavionics, 2022-09-25:
Glad you have it running now.

I saw your posting on the Ultimaker forum but I no longer post there so if you have any questions regarding my Cura builds you are better off opening an issue here and then I can respond directly to you.

Jens5395, 2022-09-25:
Thanks Mark ... and thanks for all the work you do with Cura. I hope the fact that you no longer post on the Ultimaker forum is an issue with time and not some other issue with the Ultimaker forum.