smartavionics/Cura #188: Please explain the differences in this fork.

poikilos, 2022-10-02:
The readme doesn't explain what this repo is, even though it seems to be significantly forked according to zgrosemag on Reddit: "[SmartAvionics](https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0) is a tweaked version of cura that I personally use, it's well-kept and has some interesting additions." -<https://www.reddit.com/r/3Dprinting/comments/by89xh/comment/eqjh3f9/?utm_source=share&utm_medium=web2x&context=3>

smartavionics, 2022-10-02:
https://github.com/smartavionics/Cura/releases/download/4.20.5/README.pdf

poikilos, 2022-10-02:
Since you're doing a hard fork, I don't see why you're leary of touching the main readme in the repo. Maybe at least add a link to the releases and say there is a PDF readme regarding the fork there.

kvnper, 2022-10-02:
> Thanks. Since you're doing a hard fork, I don't see why you're leary of touching the main readme in the repo. Maybe at least add a link to the releases and say there is a PDF readme regarding the fork there, if the reason you're leary is that you want to maintain all documentation in one place.

You're being a bit rude, don't you think?

poikilos, 2022-10-02:
I didn't think so, sorry. Please re-read assuming it is a suggestion.