smartavionics/Cura #189: Skin overlap not working

jmcazaban, 2022-10-19:
Hi Mark,

I am using 4.20.4, is it possible that skin overlap is broken?
It does nothing.
Using an older version it works fine.

Older version with an overlap of 200% to make it easily visible:
[image]

In 4.20.4 i get this: 

[image]

Thanks!

smartavionics, 2022-10-20:
Hi, thanks for the report. You're quite right, I goofed and broke the skin expansion. I will make a new build soon with it fixed.

smartavionics, 2022-10-20:
Hi, I've released 4.20.7 which has a fix for this issue.

jmcazaban, 2022-10-20:
Awesome, thanks for the quick fix!