smartavionics/Cura #190: Top surface skin overlap %

PhilBaz, 2022-10-21:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Cant currently control top surface skin overlap. Its just arbitrarily set.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Control of top surface skin overlap %

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Not using top surface skin, but that has other significant drawbacks.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Everyone?

**Additional context**
(Add any other context or screenshots about the feature request here.)

I thought about this before but don't think I requested it. Seems to make sense that it shouldn't just be arbitrary.

smartavionics, 2022-10-21:
UM have just made a bunch of changes to their development source that is related to the top-skin overlap. If those changes can be merged into my source, I will do that for a future release.

PhilBaz, 2022-10-21:
@smartavionics 

Thank you!

smartavionics, 2022-10-26:
Hi, I couldn't use the UM changes but have "done my own thing" and this should be fixed for the next release.

At the moment, it just uses the standard skin overlap setting, do you think it really needs a separate setting just for the top surface skins?

PhilBaz, 2022-10-26:
Ahhh, spectacular! 

Personally I do, because sometimes you want to increase to overlap to maximize strength between walls and skin (or so my thinking), yet on the top surface have it dialed down so it doesn't cause artifacts. 

In a sense one setting is functional, the other aesthetic.

smartavionics, 2022-10-26:
OK.

smartavionics, 2022-11-13:
Now in release 4.20.8.

PhilBaz, 2022-11-13:
Awesome! Thank you.

closed 2023-01-02 by PhilBaz