smartavionics/Cura #191: Dab/dots along skin edge ? Extra moves

PhilBaz, 2022-10-23:
**Application version**
(The version of the application this issue occurs with.)
latest .7

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10
**Printer**
(Which printer was selected in Cura?)
Voron 2.4
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

after laying down the initial layer there are a series of additional moves along the perimeter where the walls meet the skin.

  Ive been playing with filter out tiny gaps setting as well as fill gaps setting but cant get it to stop doing these extra dab/dots along the skin perimeter. That made me think its a bug possible?

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

File attached

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[dab dots.zip](https://github.com/smartavionics/Cura/files/9846914/dab.dots.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2022-10-23:
I can't see any problem, you'll have to provide more info, e.g. a screen shot showing the bad travels/lines.

PhilBaz, 2022-10-23:
doesn't really look like much from the travels.
[image]

PhilBaz, 2022-10-23:
Sorry, one more thing, in case I didn't state it clearly .... they are not linear moves. The nozzle dips down and makes dots a few times along the path.

smartavionics, 2022-10-23:
Ah yes, I can see now what it's doing. I think it is caused by the Monotonic Top/Bottom Order setting being enabled. Unfortunately, the UM monotonic ordering is still pretty buggy. My monotonic ordering is less buggy but, unfortunately, only works with lines skin pattern and not zigzag.

PhilBaz, 2022-10-23:
@smartavionics 

Ahhh, okay, good to know. Ill go to lines, no problems there. Thanks for the help!

smartavionics, 2022-10-23:
UM have been making changes to their monotonic code in Cura 5.x  but I don't think it's easy to retrofit those changes to my 4.20 code.

closed 2022-10-24 by PhilBaz