smartavionics/Cura #192: Stuttering/slowdown/pausing on round sections causing artifacts.

PhilBaz, 2022-10-24:
**Application version**
(The version of the application this issue occurs with.)

latest .7
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10

**Printer**
(Which printer was selected in Cura?)
Voron 2.4

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Just slice. It seems to be stuttering/slowing down/pausing on the round sections. its causing classic blob artifacts. They are even visible while playing the slice in Cura. 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[Stutter on round_Master Branch.zip](https://github.com/smartavionics/Cura/files/9852871/Stutter.on.round_Master.Branch.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2022-10-24:
You are using very small values for Maximum Resolution and Maximum Deviation. Try using larger values.

PhilBaz, 2022-10-24:
@smartavionics 

Ahhh, sorry. I am aware that can cause issues. I thought they were at default.

Ill try doubling both. or maybe going to .05.

Apologies...

PS...Very much appreciate the continued development and support of the master branch. Only version of Cura that works really ;)

closed 2023-01-02 by PhilBaz