smartavionics/Cura #194: Cooling Profile

GregValiant, 2022-11-05:
Hello,
I saw your post in UM Github ([this one](https://github.com/Ultimaker/Cura/issues/13734))regarding Cooling for ironing.  I had a similar thought but it was broader...Cooling changes at layer height and feature specific cooling (skins, outer-wall, Support Interface, etc.)
I have done this for myself coding in VB.  I don't know Python so can offer no solution.  The images below are of the dialog of my VB script.
Here it is set up for changing the fan speed at layers.
[image]

And this is for changing the fan speed for features:
[image]

IdeaMaker has a similar capability for adding cooling profiles.

I just thought I'd throw this in.  Feel free to throw it out.

smartavionics, 2022-11-05:
Hi, that looks great. Can't something similar be done as a Cura plugin? I know next to nothing about the UI so I won't be attempting that myself but I would have thought that a plugin guru would be able to insert/replace the fan commands in the gcode based on the user's settings like you show above.

GregValiant, 2022-11-05:
I know almost exactly zero Python and I have no intention of getting into it.  Old dogs and new tricks you see.
I submitted a feature request at UM Cura maybe two years ago but no one picked it up.  Since I have my own Post Processor (it reads the gcode file and then re-writes it with the changes into a new gcode file) my interest has fallen off.

5axes, 2023-01-03:
As the ironing is identified in the Cura GCode with the marker ";TYPE:SKIN" as a classic filling it is not so easy to detect it in the ISO code but it is not impossible. You need to detect two ";TYPE:SKIN" markers that follow each other.  So using a postprocessing script for that should no be difficult. Sample code made 2 years ago to add Z hop on Ironing Operation : https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/ZMoveIroning.py

I'm not using Ironing as most of my part are "technical" element for assembly  but if I remenber it was working fine.

GregValiant, 2023-01-28:
I've submitted this pull request #14394 (with help from @5axes).
It includes the "Cooling Profile" and some changes to "Insert at Layer Change".