smartavionics/Cura #195: Why no support for macOS?

meltiseugen, 2022-11-13:
Cura 4/5 offers support for mscOS. Why not this fork?

smartavionics, 2022-11-13:
Hi. I used to make builds for MacOS but unfortunately, I only have a really ancient MacBook and somewhere along the line, it became incapable of building a working Cura. I no longer need a Mac for any other reason so I won't buy one just to make Cura releases. Sorry.

meltiseugen, 2022-11-13:
> Hi. I used to make builds for MacOS but unfortunately, I only have a really ancient MacBook and somewhere along the line, it became incapable of building a working Cura. I no longer need a Mac for any other reason so I won't buy one just to make Cura releases. Sorry.

I see, if you would make cura builds for mac, I could test them for you. It's a bit hard for me to use cura in a Windows virtual machine, and I really love your features :)

smartavionics, 2022-11-13:
Sadly, the last builds I made didn't work even on systems that used to be OK. I don't know what the problem was. A further problem was that the more recent MacOS releases contained incompatible changes that were not really addressed until Cura 5 was released.

Perhaps it would be easier to install a minimalist Ubuntu 20.04 system in a virtual machine and then you could use the AppImage?

shyblower, 2022-11-13:
After cloning the current Cura repo, for my effort of porting at least some of @smartavionics's improvements over to Cura 5, I found some Github Actions in it which can generate an .AppImage for Linux an .exe for Windows and a .dmg for MacOS. These Action can be run on Github directly. They seem to set up Linux, Windows and MacOS development environments respectively and build the binaries and images. So you don't need to own any of the OSes and their build environments to build the packages.

I've only tried it once by now, without looking into the configuration and it generated at least a Windows installer .exe.
The Linux and MacOS builds threw an error, though, because of a missing certificate which is obviously needed for signing the packages. I've already created a GPG certificate for the Linux package but had no time since, to investigate this further and also will only have little time this week, because I'll be off country for 4 days. But I'll definitely try to solve this and maybe it could also be usable for @smartavionics's  Cura 4 based fork ... we'll see.

meltiseugen, 2022-11-14:
> 

This would be awesome. It would be a game changer for me

meltiseugen, 2022-11-14:
@smartavionics Could you link me the commit to the code that fixes the z-seam positioning and places the seam in a fixed vertical line please?

smartavionics, 2022-11-14:
That is all implemented in LayerPlan::addWall() but there's been so many changes in that method that there isn't a single commit that provides that capability. Look for where z_seam_point is calculated or used.

eugenmeltis3pg, 2022-11-27:
> After cloning the current Cura repo, for my effort of porting at least some of @smartavionics's improvements over to Cura 5, I found some Github Actions in it which can generate an .AppImage for [...]

Hey @shyblower ! Do you have any updates on this one? :D

shyblower, 2022-11-27:
@eugenmeltis3pg I'm working on it, but have to dive into that Conan dependency manager they now use for Cura and its components. Unfortunately the transition to Conan is still ongoing and I have to figure out quite some undocumented stuff. I'll probably invest a few hours into that today, but I can't give you an ETA when installers for my fork will be available.

meltiseugen, 2022-12-18:
Hey guys! @smartavionics @shyblower Do you have any updates on this one? :D

smartavionics, 2022-12-18:
Sorry, no news from me and I don't expect that to change as I'm not going to be looking at Cura 5.

meltiseugen, 2022-12-18:
Alright then, Ubuntu VM it is :D :D

closed 2022-12-19 by meltiseugen