smartavionics/Cura #198: Deleting some profiles seems to generate configuration errors

Jens5395, 2022-11-21:
**Application version**
4.20.8

**Platform**
Ubuntu 22.04

**Printer**
(Which printer was selected in Cura?)
Custom (???)

**Reproduction steps**
I deleted a number of unused print profiles. When I start up Cura now, it generates an error message that there are configuration errors and it then lists all the profiles that I deleted. It asks me to reset to factory defaults which would delete all printers and profiles.
If I ignore the warning message, Cura works just fine.

**Screenshot(s)**
[image]

**Expected results**
Deleting a print profile should not cause a corruption error.
(What should happen after the above steps have been followed.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
There are no recent entries in the home/.config/cura/4.20/cura.log file

**Additional information**
The error was reported in  https://github.com/Ultimaker/Cura#logging-issues but was auto-closed due to lack of further info from original poster.

(Extra information relevant to the issue.)
Please indicate what file out of all the configuration files that you need.

smartavionics, 2022-11-22:
Hi, yes, I've seen problems like this in the past. If I remember correctly, I needed to remove some other profile files that reference those names shown above. I have never understood how the Cura profiles are managed.

Jens5395, 2022-11-22:
What would reference a profile ? Why/how would one profile reference another profile? 
Very curious .....

Jens5395, 2022-11-23:
Should I maybe post this question on the main Cura forum ?

Jens5395, 2022-11-23:
I attempted to mess with the configuration files directly but ended up screwing everything up and having to restore from backup. The configuration files unfortunately make no sense. There is no 'profile' configuration subdirectory and of course there is the issue that in the Linux version at least, the configuration files are in two places at /.config/cura and at /.local/share/cura so where do you change things? I opted to change it in both locations but why have the configuration in two places to begin with ?

smartavionics, 2022-11-23:
Sorry, like I said earlier, I have never understood the Cura profile code. I can't help you. You certainly could open a similar issue on UM's repo or the forum but, as you say, they will most likely simply request you move to the latest and greatest UM Cura.

BTW, I have all of my Cura config files in git repos which from time to time I commit the latest versions and then if I really screw up the configuration I can easily get back to a previous version.

Jens5395, 2022-11-24:
Thanks anyway for replying to my predicament .