smartavionics/Cura #200: Skin Edge Support - Mission Failure

PhilBaz, 2022-11-28:
**Application version**
(The version of the application this issue occurs with.)

4.20.8

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
 win 10

**Printer**
(Which printer was selected in Cura?)

Voron 2.4

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Add skin edge support layers.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
[image]

Skin edge support is built up over the layers, but then the very last layer moves away from the top of the stack by a few line widths. It will then be an unsupported first layer and likely useless as a skin support. 

I do infill at high speed 230mm+ so it will take lots of layers of skin support to produce a reliable final layer.

**Expected results**
(What should happen after the above steps have been followed.)

The stack should be directly under the last support layer.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[Skin Edge Fail.zip](https://github.com/smartavionics/Cura/files/10100436/Skin.Edge.Fail.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

I have come to find skin edge support invaluable in getting the skin to anchor to the wall edge without missed strands. Missed strands  curl and lead to visible defects in finish, they also tend to burn and leave dark spots. After bringing this up on the Voron forum, most people seem to think they are caused by nozzle gunk, but I've looked at this closely and its missed skin curling up. It snowballs into a burnt peak.

I would love to see more control over skin edge support layers. Perhaps distance from skin edge, as well as number of walls.  Its currently also affected by 'extra infill wall count'. In the end it would be nice to control these independently. 

But most important is just getting the edge support stack to properly support itself and end under the intended target.

smartavionics, 2022-11-29:
I am looking at this, will report back when I know more....

PhilBaz, 2022-11-29:
Thank you!

I really have found skin edge support to be a critical  make or break feature.

smartavionics, 2022-11-30:
Hi, have you got any other good project files I could use to test the revamped skin edge support?

PhilBaz, 2022-11-30:
@smartavionics 

Ohh, that was quick...

Ill have a poke around and see what else is failing.

PhilBaz, 2022-12-01:
@smartavionics 

Okay, here is the result on a few test shapes....Just some whacky shapes to see the result. There are some issues. I tried to point them out in images. Ill include the images, project file and a 3mf with just the shapes. 

It seems a big pitfall of the algorithm is its desire to follow the wall shape rather than just stack directly under the target. But then I'm probably out of my element here not really understanding what is involved. 

Hope this is helpful.

PS...didn't make images for the third shape, the cylinder with fins... but it isn't really working on the whole. That being said not a great shape.

[stumpers a.zip](https://github.com/smartavionics/Cura/files/10129898/stumpers.a.zip)
[stumpers b.zip](https://github.com/smartavionics/Cura/files/10129900/stumpers.b.zip)

smartavionics, 2022-12-01:
Thanks, I'll take a look...

smartavionics, 2022-12-03:
Just released 4.20.9 that contains my rework of the skin edge support.

PhilBaz, 2022-12-03:
@smartavionics fantastic! Ill give it a go!

PhilBaz, 2022-12-04:
@smartavionics Working great. Thank you!  Saw the skin before bridge skin addition too. Fantastic!