smartavionics/Cura #201: Second/ Third skin Flow setting not respected

PhilBaz, 2022-12-08:
**Application version**
(The version of the application this issue occurs with.)
4.20.9

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10
**Printer**
(Which printer was selected in Cura?)

I was using Arachne for some TPU stuff. Found a bug on 5.2 and check if it was present on the master branch. Getting the same result with the current release.

https://github.com/Ultimaker/Cura/issues/13996

smartavionics, 2022-12-08:
Yeah, that's because your model doesn't feature a "real bridge". Instead, it has some sparse infill. That gets detected and the first skin layer above the sparse infill is printed using the bridge layer 1 settings. However, the code that handles the 2nd and 3rd bridge skins doesn't detect the sparse infill and so those skins get printed as normal skins. That's more likely to be an oversight on my part rather than a conscious decision.

PhilBaz, 2022-12-08:
Ahhh, I found the 5.2 bug on a model with a real bridge...and looks like the bug was valid with 5.2. Didnt think to double check MB with a real bridge.

Also, good to know, I would not have assumed that's the way it works. But never thought to use them till this recent TPU project.

Sorry!

PhilBaz, 2022-12-08:
@smartavionics 

Im seeing the same behaviour in 4.20.9. Maybe its a weird model?

PhilBaz, 2022-12-08:
[TPU test pieces E.zip](https://github.com/smartavionics/Cura/files/10190324/TPU.test.pieces.E.zip)

smartavionics, 2022-12-08:
I can't see that there should be any skin on that bridge you have in your picture. It's all walls.

What is weird and I can't explain is the variation in flow in the bridge walls as shown below. Your mesh isn't watertight, perhaps that's causing the weirdness?

[image]

PhilBaz, 2022-12-08:
Ahhh, yes, for skin 2 & 3 there is only a skin flow setting, not wall. I see your point. Skin 1 has wall. wasn't keeping track. 

Im going to be playing with this so ill come back with a batter question or answer. Ill recheck the model. Rhinos meshing is problematic I'm finding lately.

PhilBaz, 2022-12-08:
@smartavionics Okay a different model performs as expected.  However Rhino meshing isn't to blame. It says the mesh is closed and valid. Im saving to .3mf. No idea why its showing as not water tight.

[TPU test pieces E.zip](https://github.com/smartavionics/Cura/files/10190499/TPU.test.pieces.E.zip)

PhilBaz, 2022-12-08:
@smartavionics 

hahah, just checked and it is a proper bug in 5.2. hahah, found that one by luck!

PhilBaz, 2022-12-09:
@smartavionics 

Nope nevermind...im stupid. 5.2 works fine too. Ill close this....but if your still listening in, and god knows I don't deserve it at this point.

Mark, is there any technical reason why 'skin wall' flow is not controllable on the second and third skin?

I'm playing with TPU and bridging is a whole other level of difficult with flexibles. Its very finicky. I'm not sure if 'skin wall' flow/speed is the answer, but it would be nice to test.  To much flow just droops tremendously. And as you see in these pieces the bridging section as actually all wall.

More on that last point. All wall bridge might be desirable for TPU to get the heat of the nozzle away from the bridge as quickly as possible. I'm thinking real skin will keep concentrated heat in a given area for longer. producing more drooping.

Hope that explanation makes some semblance of sense.

closed 2022-12-09 by PhilBaz