smartavionics/Cura #202: Controllable flow and speed for 'skin wall' layer two and three. TPU help.

PhilBaz, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Mark, I'm wondering if there is a technical reason that 'skin wall' flow is not controllable on the second and third skin?

I'm playing with TPU and bridging is a whole other level of difficult with flexibles. Its very finicky. I'm not sure if 'skin wall' flow/speed is the answer, but it would be nice to test. To much flow just droops tremendously. And as you see in these pieces i was testing with bridging section is narrow enough that its actually all wall.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Controllable flow and speed for 'skin wall' layer two and three

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

echh, nothing great.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Think TPU and flexible's material users may find this helpful. I'm not sure though. testing required.

**Additional context**
(Add any other context or screenshots about the feature request here.)

More on TPU bridging. An all wall bridge might be desirable for TPU in narrow sections to get the heat of the nozzle away from the bridge as quickly as possible. I'm thinking real skin will keep concentrated heat in a given area for longer, going back and forth across the shorter length rather than quickly across the longest side in a straight line. I think more heat = more drooping. But TPU is so finicky that tuning heat lower for the bridge or other approaches will cause more problems.

I wouldn't request this for rigid material, but TPU need all the bridging help it can get.

Hope that explanation makes some semblance of sense.

smartavionics, 2022-12-09:
I have made a test release that has this new setting, please try it out...

https://www.dropbox.com/s/z9znqsmyem1oy0o/Cura-4.20.9-1-mb-amd64.exe?dl=0

PhilBaz, 2022-12-09:
@smartavionics 

Amazing.... thank you!  

Ill try it today and let you know how it goes. And if in fact this is the solution to TPU bridging.

Thank you!

PhilBaz, 2022-12-10:
@smartavionics 

Hey Mark,

Okay, firstly. I'm getting better TPU bridging from the control this offers. So that's great for anyone deep diving on TPU. 

Its working as intended. Though with some slight wonkiness--that I don't think I'm causing?

 I'm testing a plate with variations on a part. Looks like most of them have at least one flow/speed segment that is out of place. Ill attach the file so you can have a look at how its manifesting on the different shapes.  Stuff like below.

[image]

[TPU test.zip](https://github.com/smartavionics/Cura/files/10199184/TPU.test.zip)

smartavionics, 2022-12-10:
For some reason, the wall line segments generated from your models are quite short and some are shorter than your min bridge wall length setting (2mm) so they are being printed as non-bridge walls. I reduced that setting to 1mm and the lumpy bridge segments went away.

smartavionics, 2022-12-10:
The mesh is very dense, can't you optimise it to increase the length of the segments where they are in a straight line?

[image]

PhilBaz, 2022-12-10:
@smartavionics 

Okay Ill play with that....One problem though is I'm testing a model with little holes and a setting of 2mm ends up with bridging above the holes. Ill play with it though.

PhilBaz, 2022-12-10:
@smartavionics 

Yes, ill do some decimating. I like density so there is no faceting on the final models curves... this is just a test segment. The full project has round corners. But point taken.

PhilBaz, 2022-12-10:
@smartavionics 

Yes, meshing solved all the artifacts!  

 I've adjusted my mesh settings so flat surfaces aren't so dense. Thanks for the insight here. Ill use better mesh settings goin forward.

smartavionics, 2022-12-10:
Very good, thanks for the update.

PhilBaz, 2022-12-12:
@smartavionics 

Hey Mark,

After playing with this a bit more I would say the bridge wall artifacts (little flow bumps at the sides or in the middle of a bridge wall) are pretty temperamental.  Its not just a matter of small vs large segments. Its success seem overly dependent on some aspect of the meshing, but I'm not exactly sure what.  I'm trying a model with two holes and find my self going back and forth remeshing to get both of them working. One works, then I change something and the other works, but now the first is broken etc. It feels a bit buggy...

I'm attaching  a project file with 2 different relatively middle of the road meshing's. Not like the first super dense mesh I was using.
Ill also include a .step of the shape, in case you are inclined to have a go at it.

Thanks again for all you time and efforts.

[Desktop.zip](https://github.com/smartavionics/Cura/files/10204623/Desktop.zip)

smartavionics, 2022-12-12:
OK, I'll take a look when I have time, busy for a couple of days....

PhilBaz, 2022-12-15:
@smartavionics 

Thanks Mark... If you are taking another look at this and its not much trouble to add a max bridge length function it would be much appreciated. 

Thank you, thank you, thank you....

smartavionics, 2022-12-15:
I am looking at the bridging code again but I don't have any good solutions right now. I believe to fix the problems you are seeing it would need quite an extensive rework of the code that handles the bridge walls. I can't commit to doing that at the moment but will leave this open for the future.

Please explain how a max bridge length setting would behave. Does it truncate bridge wall lines when the reach the max length? Or does it completely ignore bridges that are longer than the max length? It doesn't immediately strike me as a useful capability.

smartavionics, 2022-12-15:
Actually, on further thought. I may be able to improve the current bridging logic without a major rewrite so I'll look into that...

PhilBaz, 2022-12-15:
Hi Mark,

Okay, let me know if I can help with testing if your idea works out. If its a big rework, I totally get it.

The max would not truncate. Just exclude. I find bride detection sometimes picks up what are just steep overhangs, like on chamfer/bevels, other stuff. This would be one way to try filter them out. 

Ps.....is there some way I can buy you a beer or a bag of coffee to say thanks for all your efforts, and show some gratitude?

smartavionics, 2022-12-18:
I've made some more changes and there's a new test build for you to try. It doesn't support a max bridge length setting but if you really feel you want that I can add one.

https://www.dropbox.com/s/eg2dtg59ayuuruq/Cura-4.20.9-2-mb-amd64.exe?dl=0

Thanks for the offer of goodies but I really do this stuff for fun.

PhilBaz, 2022-12-21:
@smartavionics 

Hi Mark,

Sorry, trying to wrap up work before the holiday. Ill give this a go tonight or tomorrow.

 Id just use max bridge length as more of a filter to eliminate bridges on steep overhanging walls. If you don't feel its worth the time/effort I totally get it. I wont try and sell you on it being a must have feature for every user.

Absolutely! just thought id ask in case there was already a way I didn't know about. like a 'buy me a beer' button somewhere. Well  its really amazing having a forum to directly interact and propose ideas for development and then have tools that further my work. Much like 3d printing itself this is a part of the future that blows my mind.  Thanks for indulging my ideas, and of course, cut me off like bartender when I've had to much!

Thanks again Mark for all your time and energy, not to mention your skill!

smartavionics, 2022-12-21:
I'll put in the max bridge len setting but before I do that, I need to rewrite the bridge wall code again as the recent work has taught me that it really could be done much better than it is now. I will tackle this over the next week or so.

Have a nice Christmas.

Cheers

PhilBaz, 2022-12-21:
@smartavionics 

Okay, Ill hold off testing the last version and look forward to the new one.

You as well :)

smartavionics, 2022-12-23:
Merry Christmas!

I have redone the bridge wall detection and printing code and it now supports a Maximum Bridge Wall Length setting.

Lot's of changes so I would be grateful if you can test it on various bridges and let me know if it looks wonky.

https://www.dropbox.com/s/kegv8fkbsmkdoqx/Cura-4.20.9-3-mb-amd64.exe?dl=0

PhilBaz, 2022-12-24:
@smartavionics 

Well, well..... Merry Christmas! 

Fantastic. Ill play with it tonight once the little guys are off to bed.

Hope you're having a great holiday!

PhilBaz, 2022-12-26:
Hey Mark!

Okay, I took a run with the new version. There are still some issues. Though there are fewer artifacts and many of the smaller artifacts are gone, like the little mid bridge floaters. 

Also, I'm not sure if the max bridge length is working. Or maybe I'm not understanding how it functions...  One thing I noticed is that changing it has some impact on artifacts length and location.

Ill include two project files.... One is a shape ive been playing with, another has the bridging test for the 'calibration shapes' plug in. Have a play with the pb shape project. ill also include some images in the zip.

Let me know if I can do anything else to help?

Hope you had a great Christmas!

[NBT .zip](https://github.com/smartavionics/Cura/files/10300832/NBT.zip)

smartavionics, 2022-12-26:
Thanks for testing. I have made a new build that should fix the problems with pb shape. The bridge test piece needs to be sliced with no more than 2 walls to get sensible output. You can't expect it to print walls on air.

https://www.dropbox.com/s/9gxofye93lkq20w/Cura-4.20.9-4-mb-amd64.exe?dl=0

PhilBaz, 2022-12-28:
@smartavionics 

Of course, thank you! 

Ill give it a go...... Ahh, got it!

PhilBaz, 2022-12-29:
@smartavionics 

Yes, working perfectly now!!! Awesome :)

smartavionics, 2022-12-29:
Great, I'll make a release tomorrow containing these changes.

closed 2023-01-02 by PhilBaz