smartavionics/Cura #203: A more recent Mesa version?

iandol, 2022-12-13:
I am reluctant to ask as you have generously shared your Mesa build, but have you thought to upgrade Mesa to the current stable release? Again, I fully understand that you probably are happy with 21.0.1 solving your needs and don't see a need, but just thought I'd ask!

smartavionics, 2022-12-13:
Hi @iandol , I can provide a more recent 32 and 64 bit Mesa build but I don't think it will improve the Cura display. I'll make something for the next release.

smartavionics, 2022-12-14:
I added mesa 22.2.5 builds to https://github.com/smartavionics/Cura/releases/tag/4.20.9.

iandol, 2022-12-16:
Thank you so much!!!!! Actually although I do use Cura with my Ultimaker 3, my main needs are for an up-to-date Mesa for other software. You originally made a post on the RPi forums, but that is now locked so I posted here. Thanks again!

closed 2022-12-16 by iandol