smartavionics/Cura #204: Port pressure equalizer from PrusaSlicer

meltiseugen, 2022-12-19:
**Is your feature request related to a problem? Please describe.**
PrisaSlicer gas a really cool feature that handles changes in flow/speed (e.g changing from external to internal wall)

https://help.prusa3d.com/article/pressure-equalizer-_331504

**Describe the solution you'd like**
It uses a form of pressure equalisation that is completely done in the software. It slowly decreases/increases the speed right at the end of a perimeter of the next start if another perimeter has a different speed and vice versa.

**Describe alternatives you've considered**
N/A

**Affected users and/or printers**
All

**Additional context**
https://help.prusa3d.com/article/pressure-equalizer-_331504

meltiseugen, 2022-12-27:
@smartavionics Is there any change you could look into this Prusa Slicer feature, or are you not adding any new features to the project?

smartavionics, 2022-12-28:
Hi, I took a look at that feature and I feel that it would be a lot of work to implement for little benefit.

eugenmeltis3pg, 2022-12-28:
Ah, ok then. I thought that it would have been something simpler :D. Will close this issue then. Thank you for taking a look!

closed 2022-12-28 by meltiseugen