smartavionics/Cura #210: Inconsistent Z-seam

meltiseugen, 2022-12-21:
[image]

I know that there is a feature to not place the seam in a straight line (even though the user has a fix position specified) if the wall start in the air.

But in this case It has to start in the air no matter what because there is no other option.

Is it possible to add a checkbox to disable that "don't start in the air safeguard" and keep the seam straight no matter what?

For testing, I just used a simple sphere and sliced it with user specified seam position.

smartavionics, 2022-12-21:
Use Z-seam alignment user specified and seam corner preference set to none and the seam should be dead straight.

meltiseugen, 2022-12-21:
> Use Z-seam alignment user specified and seam corner preference set to none and the seam should be dead straight.

Yep, those are the settings I used for the initial image posted

[image]

smartavionics, 2022-12-21:
OK, please supply the project file so I can test it, thanks. Do File -> Save and then either rename the .3mf file to .zip or zip it up so you can attach to this issue.

meltiseugen, 2022-12-21:
Here is the project file exported from Cura

[sphere.3mf.zip](https://github.com/smartavionics/Cura/files/10278361/sphere.3mf.zip)

smartavionics, 2022-12-21:
Ah, yes, you're right, it moves the seam point even when the seam corner preference is none. There is a small bug here that I can fix which will make it less likely that the seam will wobble but there's also a setting change that will help you. Turn off the overhang detection by setting the overhanging wall angle to 90 and the seam will  be straight.

meltiseugen, 2022-12-21:
> Turn off the overhang detection by setting the overhanging wall angle to 90 and the seam will be straight.

Ah, I understand now. Still, it would be nice to have this independently (or a way to decouple this) from the overhang setting. Basically, two of the most features that I'm interested in are the seam and the overhang controls haha

smartavionics, 2022-12-21:
It's no big deal to add a setting that would disable the moving of the z-seam when it is above air.

smartavionics, 2022-12-21:
It will be in the next release...

[image]

meltiseugen, 2022-12-22:
Good gracious, you are amazing! Thank you for listening to the community!

I'll close this one.

closed 2022-12-22 by meltiseugen