smartavionics/Cura #211: Decouple X & Y acceleration

PhilBaz, 2023-01-02:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Only in the sense that performance is not maximised.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

To adjust the acceleration for X & Y axis independently.  

I have a fast core XY. I also use input shaping ala klipper (amazing!). Input shaping suggest a shaping strategy and a max acceleration before that strategy produces 'excess rounding'. Prevailing wisdom is to respect the limit.

However the nature of corexy with the Y-axis moving 2-3x the weight of the X-axis, the input shaped acceleration limit for the X-axis is nearly 2 times that of the Y. 

Perfectly good performance rotting on the vine.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Well, the current option is to run at the low Y-axis limit. That's what I do. 

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

I guess anyone who chases performance a bit. I'm not a speed benchy guy or anything.  But I do like to run my printer at the maximum speed/reliability threshold. 

Corexy has the Y vs X axis asymmetry. Bed slingers as well. 

**Additional context**
(Add any other context or screenshots about the feature request here.)

Just an exploratorily question, as usual having no idea if this is simple or complex.  

Happy new year Mark!

smartavionics, 2023-01-02:
Have you explored using M201 to limit the acceleration for each axis?

PhilBaz, 2023-01-02:
@smartavionics 

I have not.....Ill look into that!

PhilBaz, 2023-01-02:
@smartavionics 

Hummmf.... looks like its not supported by Klipper.

PhilBaz, 2023-01-02:
Hummm.... Just reading on a klipper git requesting klipper implement this.... the response seems like a fairly definitive answer as to why its a bad idea...

"I think you guys are missing the actual reason why this is not a good idea.

Imagine trying to draw a round circle on a piece of paper, but all Y moves are slower that X moves. It would make for a horrible looking circle, and added that the Filament extrusion now needs to go "fast and slow" around this circle, Imagine what bad results that will give..."

PhilBaz, 2023-01-02:
@smartavionics Sorry to spam... yes seems to be a bit of a contentious issue if you are interested in the discussion. Looks like either way its a matter to be worked out in klipper. I think.

https://github.com/Klipper3d/klipper/issues/679

https://klipper.discourse.group/t/independant-acceleration-limits-for-x-and-y-axes/3831

closed 2023-06-01 by PhilBaz