smartavionics/Cura #214: Arc Overhang

5axes, 2023-01-06:
After the Gradient Infill the [Arc Overhang](https://github.com/CNCKitchen/arc-overhang) . An other nice Idea from CncKitchen. Would also love to test it not in so excesive conditions.  

What do you think of this  approach to print overhang ?

smartavionics, 2023-01-06:
To be honest, I'm not impressed. I think it is of limited usefulness and don't intend to implement it.

kvnper, 2023-01-08:
I just want to give credit where it's due and say it's not CNCKitchen's idea but Steven McCulloch. Thanks to CNCKitchen for bringing more attention to it.

I agree that it shouldn't be implemented yet, it's still quite new and experimental. One should wait for it to mature and see if it has any mainstay.

5axes, 2023-01-08:
Thank you @kvnper for correcting my mistake when I read CNCKitchen's readme I thought the idea came from him. I should have seen that the repository was a fork.

5axes, 2023-01-11:
Should be change into discussion not Issue

5axes, 2023-01-11:
Should be change into discussion not issue

closed 2023-01-11 by 5axes