smartavionics/Cura #217: Cumulative filament use note at layer change in gcode.

PhilBaz, 2023-02-13:
**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

a cumulative filament use note at layer change in gcode. This will help plan pauses for filament swaps at the end of the spool. 

Cant always use filament runout, as filament change on finished/top surfaces aren't good, and its nice to plan during infill.

Just a note about meters used at the end of a layer.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Echh, bad ones. 

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

I've seen there are others in an old Cura request. But I think its a feature that is generally helpful and makes sense. 

**Additional context**
(Add any other context or screenshots about the feature request here.)

Hey mark, I know you are busy for a while.

smartavionics, 2023-02-13:
Hi, yes, I'm busy but let's keep this open and I'll take a look when I can.

PhilBaz, 2023-02-13:
@smartavionics 

Thanks Mark!

smartavionics, 2023-04-30:
Hi, I've incorporated this in my latest release.

PhilBaz, 2023-04-30:
@smartavionics 

Hey, thanks so much Mark! I'm perpetually between rolls. This will help with planning. Much obliged as always!

Hope your big project went well :)

smartavionics, 2023-04-30:
> Hope your big project went well :)

Big project is still ongoing (and nothing to do with technology).

closed 2023-05-23 by PhilBaz