smartavionics/Cura #221: [FR] Discrete Line Definition Directory for Json file

5axes, 2023-03-03:
Add in the Configuration Folder a Directory to store the JSON file used to define the Discrete Line Definition : Ie

<user>\AppData\Roaming\cura\4.20\**discrete_line_definition**

Possibility to store in a dedicate folder the JSON file in order to simplify the management of the files when we use this option. 

Could be also interesting to List the files present in the Folder and give the possibility to select the file via a DropDown Control in the Cura Parameter section

smartavionics, 2023-04-30:
Hi, I've finally got around to making a release and have incorporated your suggestion. The directory is called dli_definitions.

5axes, 2023-04-30:
Thanks a lot

closed 2023-04-30 by 5axes