smartavionics/Cura #225: Hole Horizontal Expansion Max Size and Max Length seem broken

GregValiant, 2023-05-26:
**OpSys**
Windows 10 Home
**Printer**
Ender 3 Pro
**Cura Version**
SmartAvionics 4.20.1mb

**Reproduction steps**
1. The model has a vertical hole and an intersecting horizontal cross hole
2. Set HHE to 1.0 and HHE Max Size and Max Length to 0.
3. Sliced and noted the characteristic ridge above and below the cross hole.
4. Above and below the cross hole HHE appears to be properly applied
[image]

**Actual results**
1. Set the HHE Max Size to 0.1 and HHE appears to simply have turned off.
[image]

**Expected results**
HHE should be uniformly applied to the vertical hole and the "Side hole effect" should be mitigated.

**Project file**
[HHE and Max Size.zip](https://github.com/smartavionics/Cura/files/11575490/HHE.and.Max.Size.zip)

**Log file**
I don't believe this will require a log file.

**Additional information**
None

smartavionics, 2023-05-26:
It looks to me that the feature is working as intended. When HHEML is non-zero, holes whose perimeter length is >= HHEML will have HHE applied to them. Which is what your second image shows.

GregValiant, 2023-05-26:
Then I mis-understood.
So for this particular model, that rectangular cross hole is simply an interference and the only recourse is to exempt the large vertical hole from HHE...BUT...if there were holes through the bottom they could still get the HHE?

I understand now.
The bottom hole was designed at 10mm dia. and the block resting inside is 10mm cross-corner.  So the bottom hole has HHE applied and the big vertical feature is excluded.
[image]

Thank you.  I'll go ahead and close this.

closed 2023-05-26 by GregValiant

smartavionics, 2023-05-26:
> The bottom hole was designed at 10mm dia. and the block resting inside is 10mm cross-corner. So the bottom hole has HHE applied and the big vertical feature is excluded.

Yup, you got it.