smartavionics/Cura #226: Overlap settings bug when horizontal expansion is positive

PhilBaz, 2023-06-01:
**Application version**
(The version of the application this issue occurs with.)

current

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

win 10

**Printer**
(Which printer was selected in Cura?)

voron 2.4

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

When using an object to 'modify settings for overlap', if you choose 'horizontal expansion' or 'initial layer horizontal expansion' and use a positive value, the  skin layers above layer one are not affected.  This doesn't happen with negative values. 

[image]

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[Overlap bug.zip](https://github.com/smartavionics/Cura/files/11626650/Overlap.bug.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

With dual color prints adjusting horizontal expansion is useful for getting a nice distribution of initial layer lines. To eliminate thin lines or weird jaggy lines. 

Thanks again for all your work Mark!

smartavionics, 2023-06-03:
Is this also occurring with Cura 5?

PhilBaz, 2023-06-03:
@smartavionics 

Oh, I don't know.... I've almost entirely stopped using it. Its so buggy. I can check.

PhilBaz, 2023-06-03:
@smartavionics  Its bugging, but in a different way that involves bridging settings.

smartavionics, 2023-06-04:
OK, fixed this for the next release which is coming up soon (possibly today).

PhilBaz, 2023-06-04:
@smartavionics 

You're the best Mark! Thank you!

closed 2023-06-06 by PhilBaz