smartavionics/Cura #227: Zigzag for internal bridge

PhilBaz, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Internal bridge is crucial for smooth top surfaces without melted lumps or other artifacts. This might improve the reliability of internal bridges?

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Internal (and external I guess?) bridges use a line approach. But I have found if not highly tuned many of these lines will miss the edge connection and curl up, and this small issue can snowballs into a significant surface artifacts with burnt discoloration. 

I saw an image where bridging was using a zigzag pattern . This seems ideal for a few reasons, no retraction, continuous extrusion with larger edge connection contact. 

It would be great to test this option see how much difference it makes.

[image]

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

anyone using bridging

**Additional context**
(Add any other context or screenshots about the feature request here.)

smartavionics, 2023-06-02:
Hi Phil. Yes, this looks like it's worth trying so I will add settings to make the bridge skin areas use zig-zag rather than lines pattern.

PhilBaz, 2023-06-02:
@smartavionics 

Awesome!

closed 2023-06-06 by PhilBaz