smartavionics/Cura #228: 4.20.13

5axes, 2023-06-06:
I need to make a choice : use the @fieldOfView SideBar GUI Plugin or use the new function to display details of the current line in the layer view.

Without the Plugin installed 
[image]

With the Plugin SideBar GUI V4.6.1
[image]

smartavionics, 2023-06-06:
Hi @5axes , I know nothing about that plugin, perhaps @fieldofview can suggest a fix? Aldo, I anchor the (i) button to pathSlider.right.

smartavionics, 2023-06-06:
Just been thinking, a workaround could be to edit SimulationViewMainComponent.qml which will be installed somewhere in the Cura 4.20.13 tree and change line 156 from:
```
            left: pathSlider.right
```
to 
```
            right: playButton.left
```

GregValiant, 2023-06-06:
And the vertical "layer" slider gets hidden by the SideBar GUI.  That is no big thing as collapsing the Sidebar reveals the slider.  The problem 5axes brought up is a pain.
[image]
[image]

5axes, 2023-06-07:
Not a workaround
Whitout the plugin :

[image]

With the plugin 

[image]

smartavionics, 2023-06-07:
Oh, I think we need help from @fieldOfView at this point because my knowledge of qml is like < 0.

fieldOfView, 2023-06-07:
TBH, ensuring Sidebar GUI keeps working with Ultimaker Cura releases is more than enough work for me. I'm not sure if I will be able to put time into this any time soon.

5axes, 2023-06-09:
Looks like really an issue due to the plugin and not easy to solved via your code.  
`x: playButton.x - UM.Theme.getSize("small_button").width`   produce exatly the same result as `left: pathSlider.right`

The button is at the same place if the plugin is not activated but with the plugin  : 
[image]

smartavionics, 2023-06-09:
Ok, thanks for trying it. Sorry, I don't have any other ideas right now.

smartavionics, 2023-06-09:
Hi @5axes , here's another solution. The change below is applied to PathSlider.qml and it adds the line that starts Keys.onSpacePressed. With this in place, when the layer slider is focused, space will toggle the display of the line details. Hope this works for you.

```
diff --git a/plugins/SimulationView/PathSlider.qml b/plugins/SimulationView/PathSlider.qml
index 8d23c19d21..b35115e4ea 100644
--- a/plugins/SimulationView/PathSlider.qml
+++ b/plugins/SimulationView/PathSlider.qml
@@ -181,6 +181,8 @@ Item
 
         Keys.onRightPressed: handle.setValueManually(handleValue + ((event.modifiers & Qt.ShiftModifier) ? 10 : 1))
         Keys.onLeftPressed: handle.setValueManually(handleValue - ((event.modifiers & Qt.ShiftModifier) ? 10 : 1))
+        Keys.onSpacePressed: UM.SimulationView.setDisplayLineDetails(!UM.SimulationView.getDisplayLineDetails())
+
 
         // dragging
         MouseArea

```

5axes, 2023-06-10:
No change
[image]