smartavionics/Cura #229: "FIlament_Used:" line location in the gcode

GregValiant, 2023-06-06:
**Application version**
4.20.13

**Platform**
Win10 Home

**Printer**
Ender 3 Pro
**Reproduction steps**
1. Slice a model
2. Generate a gcode
3. Post process

**Actual results**
the "data" sections in the gcode start with "FIlament_Used:" instead of ";LAYER:"

**Expected results**
That the "Filament Used" lines be inserted at the end of the previous data section, or below the ";LAYER:" line.

**Project file**
N/A

**Log file**
N/A

**Additional information**
**4.20.10 snippet**
;TIME_ELAPSED:72.847665
;  Start of DATA[3]
;LAYER:1

**4.20.13 snippet**
;TIME_ELAPSED:154.089468
;  Start of DATA[3]
;FILAMENT_USED: 0.424195m
;LAYER:1

smartavionics, 2023-06-07:
As far as I'm concerned, the FILAMENT_USED comment is at the end of the previous layer.

Where does the "; Start of DATA[3]" come from? I don't recognize that.

GregValiant, 2023-06-07:
It's a little script I wrote.  It's a utility to check the data I return.

def execute(self, data):
        data[0] = ";  Start of DATA[0]" + "\n" + data[0]
        for num in range(1,len(data)-1,1):
            data[num] = "\n" + ";  Start of DATA[" + str(num) + "]" + "\n" + data[num]
        return data

[GV_AddDataHeader.zip](https://github.com/smartavionics/Cura/files/11675924/GV_AddDataHeader.zip)

GregValiant, 2023-06-09:
I've written some post processors and changed some post processors.
The new ones are:
AddCoolingProfile
AlterZHops
AddCuraSettings
BridgeTemperatureAdjustment
RemoveComments
ReNumberLayers (for one-at-a-time)
TimedCooldown
Bug fix in FilamentChange
Adjusted a couple to compensate for FILAMENT_USED line location in 4.20.13.  There were 2 or 3 posts that used "If layer.startswith(";LAYER:") which would have been false.

I don't know if you want to do anything with these, but here they are.  They work.
[GV PostProcessors.zip](https://github.com/smartavionics/Cura/files/11706019/GV.PostProcessors.zip)

Snargola, 2023-06-09:
Do these work on all versions of Cura?

GregValiant, 2023-06-09:
That was the plan.  The SmartAvionics version was the last one tested.  That's how the issue (now a non-issue) came up.

GregValiant, 2024-03-17:
In my post-processors I still have to make allowance for the the MB fork having data sections that do not start with the ";LAYER:" line.

This is the start of data[3] from a typical MB gcode file:

;FILAMENT_USED: 0.228378m
;LAYER:1
;TYPE:SUPPORT
G1 F1500 E228.3777

And this is data[3] from the same project sliced with UM Cura:

;LAYER:1
;TYPE:SUPPORT
G1 F2100 E222.45515

I always need to ensure that the code is returned in the form that is was received.
In an MB gcode, ";FILAMENT_USED:" is always the first line in a layer, not the last line of the previous layer.

smartavionics, 2024-03-17:
Greg, I don't understand the problem. Doesn't each layer start with the ;LAYER:N comment? In which case, the FILAMENT_USED line is at the end of the previous layer. Please explain what you would like it to do.

GregValiant, 2024-03-17:
Hi Mark,
This is one of those things that probably nobody else cares about.  Because I've taken to writing post processors, it is something that I always have to keep in mind, and it is something that makes Master generated gcodes different from the gcodes generated by other forks of Cura.

_"Doesn't each layer start with the ;LAYER:N comment?"_
No sir, that is incorrect.
Each layer in an MB gcode starts with ";FILAMENT_USED:".
The "LAYER:N" comment is always the **second** line in any layer of an MB gcode.
It is an easy thing to miss for anyone developing a custom Post Processing Script, and can cause trouble when trying to use a post processor developed for any other Cura fork  - with MB Cura.

I searched through your "Cura" code and through your "CuraEngine" code and was unable to find where ";FILAMENT_USED:" was being added/inserted.  If you can point out where that is occurring, maybe I can put together a PR.

smartavionics, 2024-03-17:
Take a look at the mb-master branch of CuraEngine. It's in writeLayerComment() which is in gcodeExport.cpp. It currently looks like this...
```
void GCodeExport::writeLayerComment(const LayerIndex layer_nr)
{
    const Scene& scene = Application::getInstance().current_slice->scene;
    *output_stream << ";FILAMENT_USED: ";
    for (size_t i = 0; i < scene.extruders.size(); i++)
    {
        if (i > 0)
        {
            *output_stream << ", ";
        }
        if (flavor != EGCodeFlavor::MARLIN_VOLUMATRIC)
        {
            *output_stream << getTotalFilamentUsed(i) / (1000 * extruder_attr[i].filament_area) << "m";
        }
        else //Use volumetric filament used.
        {
            *output_stream << getTotalFilamentUsed(i) << "mm3";
        }
    }
    *output_stream << new_line;
    *output_stream << ";LAYER:" << layer_nr << new_line;
}```

That function is called at the start of each layer.

GregValiant, 2024-03-17:
Geez, I'm barely conversant in Python.  I'll take a shot at it though.

In that function:
*output_stream << ";FILAMENT_USED: ";
comes before
*output_stream << ";LAYER:" << layer_nr << new_line;

Should it be:
void GCodeExport::writeLayerComment(const LayerIndex layer_nr)
{
    const Scene& scene = Application::getInstance().current_slice->scene;
    ***output_stream << ";LAYER:" << layer_nr << new_line;**
    *output_stream << ";FILAMENT_USED: ";
    for (size_t i = 0; i < scene.extruders.size(); i++)
    {
        if (i > 0)
        {
            *output_stream << ", ";
        }
        if (flavor != EGCodeFlavor::MARLIN_VOLUMATRIC)
        {
            *output_stream << getTotalFilamentUsed(i) / (1000 * extruder_attr[i].filament_area) << "m";
        }
        else //Use volumetric filament used.
        {
            *output_stream << getTotalFilamentUsed(i) << "mm3";
        }
    }
    *output_stream << new_line;
}

Also - is "MARLIN_VOLUMATRIC" a typo?  Should it be "VOLUMETRIC"?  It wouldn't matter so long as it was consistent throughout the code.

smartavionics, 2024-03-17:
I don't think so because the reported filament used is for the previous layer, not the new layer.

Yes, volumatric is most likely a typo but it has been like that for many years.

GregValiant, 2024-03-17:
Would this move it to the end of the correct layer (or would the volume be incorrect?)
A layer should begin with ";LAYER:" and end with ";TIME_ELAPSED:"

(Line 527)
void GCodeExport::writeTimeComment(const Duration time)
{
	const Scene& scene = Application::getInstance().current_slice->scene;
    *output_stream << ";FILAMENT_USED: ";
    for (size_t i = 0; i < scene.extruders.size(); i++)
    {
        if (i > 0)
        {
            *output_stream << ", ";
        }
        if (flavor != EGCodeFlavor::MARLIN_VOLUMATRIC)
        {
            *output_stream << getTotalFilamentUsed(i) / (1000 * extruder_attr[i].filament_area) << "m";
        }
        else //Use volumetric filament used.
        {
            *output_stream << getTotalFilamentUsed(i) << "mm3";
        }
    }
    *output_stream << new_line;	
    *output_stream << ";TIME_ELAPSED:" << time << new_line;
}

smartavionics, 2024-03-17:
Ah, so you're saying that a layer ends with TIME_ELAPSED and that FILAMENT_USED must come before that?

OK, I'll make that change, don't worry about creating a PR.

GregValiant, 2024-03-17:
Thanks.
It's a consistency thing.

While fooling around, I came across the "VOLUMATRIC" typo discussions.  It's actually kind of funny.

GregValiant, 2024-03-17:
BTW I'll go ahead and close this.
Thanks again.

closed 2024-03-17 by GregValiant