smartavionics/Cura #235: [4.20.15] Layer view antialiasing

5axes, 2023-11-26:
Windows Version: Windows 10 (version 18363), édition 64-bit

Graphics Quatro RTX  3000
GL version: 4.6.0
Vendor: NVIDIA Corporation

In my configuration **Enable Layer Antialiasing**  generates Aliasing. Can you explain when activating this option?

Enable Layer Antialiasing  generates Aliasing : **OFF**
[image]

Enable Layer Antialiasing  generates Aliasing : **ON**
[image]

smartavionics, 2023-11-26:
I know the above example is trivial but could you please provide a project file so that I can see if I get the same visual effect with exactly the same model and settings. Thanks.

5axes, 2023-11-26:
No problem but as you says it's very trivial.
**Off**
[image]
**On**
[image]

[Cube-visu.zip](https://github.com/smartavionics/Cura/files/13468418/Cube-visu.zip)