smartavionics/Cura #237: 4.20.16 Can You Explain ?

5axes, 2023-12-17:
Sorry but I don't really see the benefits of this nez release. I made the test with this part  :https://github.com/Ultimaker/Cura/issues/17208

And the orientation doesn't change ?  Should we have to activate a new parameter ?
[image]

[Cable_Keeper_4.20.16.3mf.zip](https://github.com/smartavionics/Cura/files/13696579/Cable_Keeper_4.20.16.3mf.zip)

smartavionics, 2023-12-17:
Ah, I was just wondering about that part, I am mistaken in that it was already being handled OK. I will remove the reference to that issue from the release info.

The change I have made is that now when it is deciding how to orientate the bridge skin lines it uses the heuristic that the lines will be at 90 deg to the longest unsupported edge that the skin has unless the longest supported edge is > 3 times the length of the longest unsupported edge in which case the lines will be parallel to the longest supported edge.

Previously, it always based the angle on the direction of the longest unsupported edge without taking into consideration the supported edges.

The factor of 3 is hard coded and I would prefer not to have yet another setting for it. Maybe that will be required if it gives poor results.

I have collected together various models that all have bridges and they all now slice (IMHO) correctly...

[image]

smartavionics, 2023-12-17:
Note! I just edited the previous post as I had the description topsy-turvy!

5axes, 2023-12-17:
Not totally convinced of the result... On the second example : https://github.com/Ultimaker/Cura/issues/16917
**Version 4.20.15**
[image]

**Version 4.20.16**

[image]

I don't see any difference ?

Personally, I'd prefer it the other way round... Lengthwise for the smallest zone, widthwise for the largest zone.

smartavionics, 2023-12-17:
That's interesting. What layer height are you using? I'm using 0.2 and I get this...

[image]

5axes, 2023-12-17:
I've loaded the 3mf file of the example PR .. Same as you layer height 0,2 
[image]

[irrational Bridging 4.20.16.3mf.zip](https://github.com/smartavionics/Cura/files/13696843/irrational.Bridging.4.20.16.3mf.zip)

smartavionics, 2023-12-17:
Check the bridge skin overlap, for this model it has to be > 0%, even 5% is sufficient. It's a weird model because of that radiused edge under the bridge. Without the overlap, that long edge becomes unsupported so the skin lines are orientated parallel to it.

5axes, 2023-12-17:
Bridge Skin Overlap  must be at least 0,5mm to get the result.

[image]

smartavionics, 2023-12-17:
With my settings, I don't need so much...
[image]

techflick, 2024-05-10:
Hi, I'm the Author of the Issue https://github.com/Ultimaker/Cura/issues/16917 for now it seam to be resolve. But as I tried to follow this thread I realised that I can not find the setting to play through the different cases, has the bridge skin overlap settings bin removed?

smartavionics, 2024-05-10:
Hi, the bridge skin overlap setting isn't provided by UM Cura, just my variant.