smartavionics/Cura #238: Bridge on longest edge for internal bridging...option?

PhilBaz, 2023-12-18:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Yes. missed connections at the edge of an internal bridge often become the cause of a surface artifact. the missed connection curls, then chars, then snowballs into a full on surface bump.

It seems the fewer edge connections the better. 

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

The option to use the longest edge for internal bridges. to minimise connections at the edge.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Its really software dependent. Would just like an option to control.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

All.

**Additional context**
(Add any other context or screenshots about the feature request here.)

The powers that be sometimes indulge my requests. I have tested this extensively because the product i print over and over has large flat surfaces that are easily ruined by internal bridging issues. Perhaps this is also more of an ABS issue. That's the material Im using most.

Thanks for continuing to support the best slicer available! Happy holidays Mark!

smartavionics, 2023-12-18:
Hi @PhilBaz , what do you mean by internal bridges? Are they bridges where all edges are supported?

Surely there has to be a limit to how long the bridge skin lines should be? It seems to me that you're suggesting that a bridge that was, say, 10cm long and 1cm wide should be printed with 10cm long skin lines as opposed to 1cm skin lines which is how the current code would work.

Doesn't the bridge skin overlap help with attaching the skin lines?

PhilBaz, 2023-12-18:
@smartavionics 

Hey Mark,

Ahh, of course, I mean bridging over infill.  I set the bridge sparse infill max density very high so I am always controlling first layer over infill. I know Cura default is to just treat it like skin, but i find better results with a bridging approach, and i've seen prusa derived slicers take this approach by default as well. 

I have found this leads to increased quality and repeatability of surface skin layers.

But also im mostly using ABS, where you want to keep fan low to avoid warping parts. On large parts even high fan on one layer can do it.