smartavionics/Cura #241: [4.20.16 ARM64] Drag and drop not working on Wayland

Botspot, 2024-01-21:
**Application version**
4.20.16 ARM64 AppImage

**Platform**
Raspberry Pi OS Bookworm, using Wayfire with known working Xwayland, RPi5.

**Reproduction steps**
Drag a STL file onto the print area from a file manager of your choice.

**Actual results**
Nothing happens.

**Expected results**
The model would appear on the print area.

Clicking File -> Open in Cura does successfully open an STL file, so this seems to just be an issue with drag and drop.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
This logging-issues link you put in the issue template is no longer leading to any useful logging information. You likely need to change the link to an earlier version relating to Cura 4.

**Additional information**
This worked fine on PiOS Bullseye which used Xorg.

smartavionics, 2024-01-21:
Yeah, a quick google shows that drag and drop to an X11 application running under Wayland is unreliable or completely broken. If I become aware of a way of fixing this I will give it a go but, otherwise, I'm not going to spend any time on this issue.

Botspot, 2024-01-21:
That is understandable. But I was under the impression that Cura was capable of running on Wayland natively, as per the release notes. Is that not the case? Is it possible to force Cura to use wayland by setting some environment variables?

smartavionics, 2024-01-21:
Ah, yes, the release notes state that setting QT_QPA_PLATFORM=wayland would make it use native Wayland. That currently only works (but not perfectly) on the x86 build. I will try and make an aarch64 build that can do the same. I'll get back to you when I know more.

smartavionics, 2024-01-22:
I rebuilt Qt with Wayland support and the resulting Cura appimage will use Wayland. Unfortunately, it has some showstopping problems. Menus don't work right in that if you click on any menu heading other than File (i.e. Edit, View, etc.) the menu pops up but then disappears to be replaced by the menu on the left and that is subsequently replaced until you end up with the File menu displayed every time. Another problem is that drag and drop now works for stl files but if you drop a 3mf file onto Cura it crashes due to some internal Qt error. These problems could possibly be fixed by using a more recent version of Qt so my next step is to try that.

Botspot, 2024-01-22:
Thanks for the work on this. I presume a Wayland build would give a nice graphics speed boost out-of-the-box for PiOS Bookworm users, so if this can work, it would benefit everyone.

smartavionics, 2024-01-22:
Today's update... I built a version using Qt 5.15.12 and that has fixed the problem with the menus. It still crashes when drag/dropping a .3mf file but .stl and .obj files are OK. I have therefore disabled drag/drop for .3mf files. Not a great solution but it's the best I can do for now. I have uploaded a test for you to try out. It's at https://github.com/smartavionics/Cura/releases/download/4.20.16/Cura-4.20.16-mb-aarch64-16k-wayland-test.AppImage

As for performance, Wayland doesn't provide any benefit over X11 as far as I can tell. The pi5 is noticeably faster than the pi4 so I have been experimenting with different shaders for the layer view. Hitting Ctrl-space in the layer view toggles use of a shader that's a bit more like the shader used for x86 builds. It looks a little nicer as it fills the gaps between line segments but does incur a performance hit so it's nice to be able to toggle it.

Compare these images to see the visual difference it makes...

[image]
[image]

Botspot, 2024-01-22:
Thanks for the appimage to test out. I can confirm drag-and-drop works. One minor regression I noticed is that it seems to be using fewer possible colors.
Compare these two screenshots and notice that the top one (wayland) has rings of yellow color bands on the rendered model, and blocky grey rectangular chunks in the gradient at the top for the [Prepare Preview Monitor] tabs.
[image]
[image]
Could this be due to the fact that My Pi5 is using the 4k kernel? I disabled the 16k kernel, in part because Cura would not launch on the 16k kernel a few months ago.

Botspot, 2024-01-22:
Regarding shaders, I have never actually gotten the enhanced layer view working.
On Bookwork, I just ensured that `libgles-dev` and `mesa-utils` were installed, then ran your wayland test appimage again and the layers still look like this: 
[image]

smartavionics, 2024-01-22:
Are you sure you haven't enabled the force layer view compatibility option in the general preferences?

As for the prepare view images, they look the same to me?

Botspot, 2024-01-22:
> Are you sure you haven't enabled the force layer view compatibility option in the general preferences?

That was, in fact, the case. Enabling the normal viewing mode solved that. With Ctrl+Space, I can see the performance difference between the 2 shaders. If you start going forward with releasing Wayland builds, could you change the default to use the faster shader with less polygons? On larger models with many curves I found this makes a big performance impact with a near zero visual difference.

> As for the prepare view images, they look the same to me?

Two separate areas have gradients: the flat face of the model and the UI background surrounding the [Prepare Preview Monitor] buttons. Notice how that grey band across the entire width of Cura is more blocky in my wayland screenshot when compared to your Xorg build. Something is reducing the number of used colors.
For the flat face of the model, this is harder to see on some monitors, so I have used kolourpaint to flood fill all pixels of an exact color for comparison. Thicker bands means less colors are used and the gradient is more "chunky".
Wayland:
[image]
X:
[image]

smartavionics, 2024-01-22:
Ah, yes, I can see that effect now on my own models. Haven't a clue as to what's causing that.

BTW, the test appimage defaults to using native Wayland but you can run that same test appimage using X11 rather than Wayland by using this command line:

(unset WAYLAND_DISPLAY;./Cura-4.20.16-mb-aarch64-16k-wayland-test.AppImage)

Botspot, 2024-01-22:
I am somewhat surprised that a build for 16k is running fine on my 4k kernel. Any thoughts? If this works everywhere, then would there be any reason to maintain 2 builds?

smartavionics, 2024-01-22:
I need to keep the other builds because an appimage built on bookworm won't work on older systems. My build systems are still running buster.

smartavionics, 2024-01-23:
I have fixed the loss of colour resolution, it took me almost the whole day so I hope it is appreciated.
I have updated the test build so could you please download it again and give it a go.

Botspot, 2024-01-23:
> I have fixed the loss of colour resolution, it took me almost the whole day so I hope it is appreciated. I have updated the test build so could you please download it again and give it a go.

Test build works great. Out of curiosity, what did end up fixing the loss of color resolution? And could this be a general issue that other projects could encounter when switching to wayland?
Will this build work on Xorg fine, or will we need to adjust our installation script to install the wayland appimage on wayland distros?
Also, just wanted confirmation that you changed the default layer view shader to the faster one? It seems that you did, but I just wanted to make sure.

smartavionics, 2024-01-24:
Test build works great. Out of curiosity, what did end up fixing the loss of color resolution?I now explicitly request 8 bits for each colour. What I don't understand is why under X11 it has always worked without doing that. And could this be a general issue that other projects could encounter when switching to wayland?I don't think so.
Will this build work on Xorg fine, or will we need to adjust our installation script to install the wayland appimage on wayland distros?It will be ok for 64 bit bookworm systems using either Wayland or X11.
Also, just wanted confirmation that you changed the default layer view shader to the faster one? It seems that you did, but I just wanted to make sure.Yes, it defaults to the faster shader

Botspot, 2024-01-24:
@smartavionics, could you go to github and check the response to this issue you just added via email? I think some of your formatting was not intentional.

closed 2024-02-11 by Botspot