smartavionics/Cura #243: Strange flow changes in vase mode.

bartwaw, 2024-02-10:
**Reproduction steps**
1.Attached file.
2. Vase mode

**Screenshot(s)**
([image].) 

[image]

**Actual results**
[image]

[image]

**Expected results**
(flow should be same.)

**Project file**
(attached)

smartavionics, 2024-02-10:
Please provide the project file, thanks.

bartwaw, 2024-02-10:
[CFFFP_clip-smal-2l.zip](https://github.com/smartavionics/Cura/files/14230551/CFFFP_clip-smal-2l.zip)

bartwaw, 2024-02-10:
Mark, you're probably going to get tired of me soon, because as you can see, I'm starting to search for errors and look for new solutions :)

smartavionics, 2024-02-10:
Looking at your project file, I can see that you have a solid model but then you have set the minimum spiralized line width to < 100%. The variable line width functionality only works with non-solid models. If you set it to 100% you don't get the flow anomalies.

bartwaw, 2024-02-10:
Is possible that this settingls will work with solid ? 
 Generally is easier to design models for vase mode that are solid. and specially that walls will need to meet each other. As you see it's clip for stuff. To get better strength some layers should stick to other layers. 

[image]
[image]

bartwaw, 2024-02-10:
checked, not solid model gives even worst effects:
[image]

[image]

bartwaw, 2024-02-10:
But strange. 
smaller version gives such results:
[image]

but almost the same, but taller looks almost good:
[image]

[CFFFP_clip-tiny-35mm tall.zip](https://github.com/smartavionics/Cura/files/14230660/CFFFP_clip-tiny-35mm.tall.zip)

smartavionics, 2024-02-10:
Can you provide the non-solid model, please?

bartwaw, 2024-02-10:
[clip-tiny-35mm tall- not solid.zip](https://github.com/smartavionics/Cura/files/14230666/clip-tiny-35mm.tall-.not.solid.zip)

smartavionics, 2024-02-11:
I've fixed some bugs and it's looking more sensible. Here's your not-solid clip. You can see from the info box at the bottom that the line width is 0.86 which I think is close to the width of the walls in the model. I will make a replacement 4.20.17 release tomorrow so you can test it.

[image]

bartwaw, 2024-02-11:
Mark, thanks for your effort. 
But as i understand well, "minimum spiralized line width" works only  when object is not solid. Correct ?
As i said is easier to design such objects as solid - so it's possible to that feature will work for solid objects ? Just asking. 
Or maybe, just thinking, Arachne engine will "hold" better such features - smooth start and end of spiralized contour to avoid squishing at begining and at the end. I know that you don't plan using arachne and maybe i should ask at cura githib to such improvements.

smartavionics, 2024-02-11:
Hi, you can use the variable line width feature with solid objects but that is not the intended use. So here's your solid model sliced using a max spiralized line width of 150%. You can see that the lines are fat and they actually overlap at the hinge point.

[image]

You can request the tapered start/stop of the spiral on Ultimakers github but I very much doubt they will do it. I offered it to them years ago but they weren't interested then and since that time they broke their spiralization by stopping it working with non-solid models.

smartavionics, 2024-02-11:
I will make a new build for you today, what system? Windows?

bartwaw, 2024-02-11:
Yes, windows.

smartavionics, 2024-02-11:
I've updated the Windows build to incorporate the bug fixes mentioned above.

https://github.com/smartavionics/Cura/releases/download/4.20.17/Cura-4.20.17-mb-amd64.exe

smartavionics, 2024-02-11:
Here's another change that isn't in the build I posted an hour or so ago but I think it is needed. When you have solid models and the model is narrower than twice the max line width, the walls get squished together which will cause over extrusion. So I am proposing to limit the wall width in that situation as shown below. I've cranked the max line width up to 150% to make it obvious.

[image]

smartavionics, 2024-02-11:
I have updated the builds again to incorporate the wall width limit mentioned in my last comment.