smartavionics/Cura #249: Cura 4.20.18| There is an invalid empty run in the bottom layer print path of the print file.

Eazao, 2024-04-19:
We mainly do ceramic 3D printing, there is a problem with the bottom layer of printing, as shown below, please check the file.

Hi [smartavionics](https://github.com/smartavionics),
The red line in the picture is the scratch caused by the empty path of cura 4.20.18 version. How to close it?
cura 4.20.18

[image]

The picture below is the path of Cura 5.7, without this scratch.
Cura 5.7
[image]

This is the project file:

[cup.zip](https://github.com/smartavionics/Cura/files/15039825/cup.zip)

smartavionics, 2024-04-19:
Hi @Eazao , that file isn't a project file because it contains no settings. Please do File->Save (not Export). Thanks.

Eazao, 2024-04-19:
> Hi @Eazao , that file isn't a project file because it contains no settings. Please do File->Save (not Export). Thanks.

Yes, we exported the project according to your instructions.
[cup.zip](https://github.com/smartavionics/Cura/files/15041438/cup.zip)
[cup (2).zip](https://github.com/smartavi
[cup (3).zip](https://github.com/smartavionics/Cura/files/15041445/cup.3.zip)
onics/Cura/files/15041442/cup.2.zip)

smartavionics, 2024-04-20:
Hi @Eazao , could you please upload the gcode file that produced the scratches. I can't reproduce the problem here. Thanks.

smartavionics, 2024-04-20:
Don't worry about the gcode, I found the ufp file. Thanks.

smartavionics, 2024-04-22:
Hi, I looked at the gcode in the UFP file and it looks like your second photo, not the first. Are you sure you have the photos correctly matched to the slicer version?

smartavionics, 2024-04-22:
Here's how the 4.20.18 gcode looks...

[image]

Eazao, 2024-04-23:
Hi [smartavionics](https://github.com/smartavionics),

Thank you for your reply.

We have just re-sliced and printed this gcode file and shot the video at the same time. Please check the latest problem, which is the location of the red line mark in the picture.
[image]

[heige_cup.zip](https://github.com/smartavionics/Cura/files/15073781/heige_cup.zip)

[heige_cup_1.zip](https://github.com/smartavionics/Cura/files/15073778/heige_cup_1.zip)

smartavionics, 2024-04-23:
Hi, thanks for the extra evidence. Yes, it's a problem with the combing mode "not in skin", use another mode or no combing. I will look into fixing this for the next release.

smartavionics, 2024-04-23:
I have found the bug and it will be fixed in the next release. Thanks for the detailed report.