smartavionics/Cura #250: 3D Honeycomb

5axes, 2024-05-03:
As "Infill pattern lower" you should love to integrate this 3D Honeycomb pattern.

[image]

Original source : https://github.com/SoftFever/OrcaSlicer/pull/4425  
from @gringer

smartavionics, 2024-05-04:
Yes, it does look interesting. I looked at the code they use in OrcaSlicer and some day I may rewrite it for Cura. Too busy at the moment, though.

gringer, 2024-05-05:
I already submitted code for Cura many years ago; it was rejected.

smartavionics, 2024-05-05:
Thanks for the offer of help. I would be very happy to take something that embodied the algorithm and merge it into my Cura but I no longer submit PRs for new functionality to the upstream repo.

smartavionics, 2024-05-05:
So, I've literally just cut and pasted the code in your PR and it compiles and runs in my Cura. However, it doesn't look like the animation shown above as it is filling in horizontal squares...

[image]

smartavionics, 2024-05-05:
OK, I actually looked at the code and it's easy to disable those filled in regions.

https://github.com/smartavionics/Cura/assets/585618/d6e7ee43-b652-4654-86bd-a54c1b336e23

smartavionics, 2024-05-05:
However, it is still duplicating a lot of the lines. Do you have any suggestions as to how that can be fixed? Thanks.

smartavionics, 2024-05-05:
I've managed to fix the problems found, so far...

[image]

gringer, 2024-05-06:
That looks awesome, well done.

Sorry about the delay in response; timezones don't really work in my favour.

gringer, 2024-05-06:
FWIW, I also did a bit of fiddling with line separation to fix density issues, trying to get the actual amount of extruded filament as close as possible to that of the rectilinear infill:

https://github.com/prusa3d/PrusaSlicer/pull/6434/commits/36ad27c49104fbd576277334f2fc38ea49fe012b

Not sure if this would be of interest to you. I only bothered with it because Stefan brought it up in one of his testing videos, and it was mentioned in a PrusaSlicer bug report.

smartavionics, 2024-05-06:
Thanks, I have tweaked the density and also the Zscale computation which is currently lineSpacing * 1.85 (rather than sqrt(2)) as this gives more uniform shaped octagons. That was done by trial and error and I don't understand what the 1.85 relates to. Anyway, it's looking pretty good and will be in my next release.

PS I noticed when printing some sparse 3D honeycomb infill without walls that I was getting underextrusion following long travel moves so I have introduced a couple of settings that let you set travel distance proportional prime amounts at the start of infill lines. That completely fixes the underextrusion.

smartavionics, 2024-05-14:
BTW, 4.20.20 has been released which includes the Honeycomb 3D infill pattern.

closed 2024-05-14 by 5axes