smartavionics/Cura #251: gcode files

SirGit1646, 2024-05-20:
from Cura 4.20.... on Cura can't read gcode files and can't import settings from .gcode file,  I hope someone can help
thanks

smartavionics, 2024-05-20:
Hi, please provide an example file that it cannot read.

SirGit1646, 2024-05-21:
the one with the .gcode filetype are Cura 4.13

[image]
[image]
[image]
[image]

smartavionics, 2024-05-21:
Thanks for the images. It works for me as you can see in my image below. Could you please upload a zipped cura.log file so I can check if there are any errors reported. Thanks.

[image]

SirGit1646, 2024-05-21:
[image]
just rename it
regards

SirGit1646, 2024-05-21:
Amazing.........what OS you are using  , btw. I'm using Raspberry pi 5 and 4  with Debian Bookworm . I don't have any  log files tell me where to find

smartavionics, 2024-05-21:
Do Help Menu, Show Configuration Folder and it will open the file manager on the directory that contains cura.log. Zip or rename and upload, please.

My images are from a pi4 running Bookworm.
[image]

SirGit1646, 2024-05-21:
thanks I found it
[4.20.zip](https://github.com/smartavionics/Cura/files/15389439/4.20.zip)

SirGit1646, 2024-05-21:
yeah I could see gcode reader are in all of my copies disabled

smartavionics, 2024-05-21:
Yes, your log file shows them as disabled...
```
2024-05-21 00:32:19,157 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [428]: Plugin [GCodeGzReader] has been disabled. Skip loading it.
2024-05-21 00:32:19,172 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [428]: Plugin [GCodeGzWriter] has been disabled. Skip loading it.
2024-05-21 00:32:19,187 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [428]: Plugin [GCodeProfileReader] has been disabled. Skip loading it.
2024-05-21 00:32:19,212 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [428]: Plugin [GCodeReader] has been disabled. Skip loading it.
```

SirGit1646, 2024-05-21:
I download 4.20.20 appimage  yesterday, could you edit your appimage files
best regards
Sigi

smartavionics, 2024-05-21:
I do not disable those plugins, it must be your configuration that is disabling them.

SirGit1646, 2024-05-21:
thank you sooo much I never did know there is a plugin to enable, because I never disabled anything in marketplace
I only installed some plugins
best regads
Sigi

smartavionics, 2024-05-21:
I can't explain how they became disabled, but at least the issue is solved now.