smartavionics/Cura #253: incorrectly set temperatures for T0 and T1

Rwide, 2024-06-28:
**Application version**
4.20.21

**Platform**
WIndows 11

**Printer**
Custom printer, Snapmaker J1

**Reproduction steps**
1. import 2 objects. select extruders for them.
2. The temperature for extruder 1 is set to the temperature of extruder 2.
3. Not sure what happends after toolchange to extruder 2, it seems to be a mess, but i believe the temp for extruder 1 was set to normal temperature instead of standby temperature..

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2024-06-29:
I have no experience with multi-extruder printers but if you provide a simple project file that shows the problem, I will investigate.

Rwide, 2024-06-29:
Hello. I made a mistake. I was trying to create a post proccessing script for conditional dual purge lines for my dual extruder printer yesterday and I must have accidently switched to another version of Cura while testing. The behaviour described is not present in your version. Sorry =). 

It seems to me that in order to make such a post proccesing script work in your fork, one would need a few more data points in the generated gcode. Cura 5.7 presents those data points but unfortunately has the described behaviour above =D

I'm reffering to these:
Printing temperature for extruder T0
Printing temperature for extruder T1
Standby temperature for extruder T0
Standby temperature for extruder T1

Would it be possible to add that functionality?

smartavionics, 2024-06-29:
Sounds easy enough. Can you provide me with an example of gcode that shows exactly what you are expecting to see. Just a few lines is sufficient. Thanks.

Rwide, 2024-06-29:
Sure! Cura 5.7 does not present the standby temp unfortunately, but it does present the printing temperature and other data like this:

;Header Start
;Version:1
;Slicer:CuraEngine
;Printer:Snapmaker J1
;Estimated Print Time:11147
;Lines:85239
;Extruder Mode:Normal
;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:PLA
;Extruder 0 Print Temperature:220
;Extruder 0 Retraction Distance:1
;Extruder 0 Switch Retraction Distance:1
;Extruder 1 Nozzle Size:0.4
;Extruder 1 Material:PLA
;Extruder 1 Print Temperature:260
;Extruder 1 Retraction Distance:1
;Extruder 1 Switch Retraction Distance:1
;Bed Temperature:60
;Extruder(s) Used:2

In your fork, the temperatures are presented at the bottom, but there is nothing that specifies wich extruder is wich..

Best regards

smartavionics, 2024-06-29:
I can't find where the above data is generated in the UM source. Could it be being generated by a plugin?

smartavionics, 2024-06-29:
I found where that data is coming from, it's the snapmaker plugin.

Rwide, 2024-06-29:
Ah I see, sorry about that =)

After deleting the plugin, there is less data in generated gcode. The remaining data looks like this:

;FLAVOR:Marlin
;TIME:5568
;Filament used: 22.0569m, 0m
;Layer height: 0.2
;MINX:87.205
;MINY:25.201
;MINZ:0.28
;MAXX:236.792
;MAXY:174.795
;MAXZ:10.08
;TARGET_MACHINE.NAME:Snapmaker J1
;Generated with Cura_SteamEngine 5.7.2

So at least there is one "normal setting" left (layer height). =)

smartavionics, 2024-06-29:
I've started working on getting the temperatures output, here's an example (single extruder used)...
```
;FLAVOR:Marlin
;TIME:723
;Filament used: 0.157907m
;Layer height: 0.2
;MINX:113.313
;MINY:102.445
;MINZ:0.2
;MAXX:166.37
;MAXY:137.552
;MAXZ:10
;LINEARADVANCEPROCESSED
;Generated with Cura_SteamEngine master
M140 S45
M105
M190 S45
;Extruder 0 Print Temperature:200
;Extruder 0 Standby Temperature:180
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
```
Do you need the lines in a particular place or require any header or footer around them?

Rwide, 2024-06-29:
It looks good as it is. I (chat gpt) can adapt to the situation. I don't do much coding myself, but I have found that chat GPT, if given well described descriptions of intended functionality, parameters, suggested  logic and attempts can do wonders. =D

Rwide, 2024-06-29:
A datapoint such as "extruders used: 1,2" would solve that problem if it arises..