smartavionics/Cura #255: Monotonic Zig-Zag possible?

PhilBaz, 2024-07-07:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Yes. As I go faster and faster i've now realized I can lay down first layers of ABS apraching 300mm^2 at 35mm^3 +. I had arbitrarily stopped my self at 100mm^2 at some point for no reason.

This has been a huge revelation, as slow first layers cause a number of problems with enclosed expanding printers by greatly expanding the time between first and second layer and thus thermal expansion between layers.

Anyhow that is all to say, It seems to me I need to use zig-zag at those speeds to minimize retractions which seem to cause a problem. I need to test this more but the retractions cause tinny missing extrusions latter down the line. Zig-zag appears to solve this problem. 

But.....now its back to square one with the absence of monotonic and slight variations in the layer due to the...non-monotonicness (?)

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Monotonic zig-zag....please.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Ill need to test more as mentioned above. But highspeed first layers are a real benefit... and if monotonic--with all its retractions-- causes extrusions issues as appears to be the case with initial tests then it won't be suitable. 

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Fast ones....

**Additional context**
(Add any other context or screenshots about the feature request here.)

Hello Mark! Hope all is well your side of the pond :)  Thanks again for continuing to support the best version of Cura. Coincidently the only one on my computer. I still find UM Cura to troublesome and have stopped following its development.  I use your Cura for advanced projects requiring maximum control and Orca for little or quick things just because it's UI and workflow do offer some very nice features, thought the level of control is lacking.

One large difference I have noticed between Cura and Orca.....Curas first layers are much nicer. Like by a lot. Not sure why. By comparison Orcas first layers (real print, not slice, and with identical settings) are filled with artifacts and issues at the borders where skin meets walls. even on simple squares etc. Its very noticable. I only mention it because its such a big difference I have not seen mentioned and its a real drawback. 

Anyhow....thanks again for everything. Thought I would write with this request see if it's something you might consider.

smartavionics, 2024-07-07:
Hi Phil, thanks for the kind words. Now, as for monotonic zig-zag, I will look into being able to do that. I remember that at the time I implemented the monotonic ordering that I thought that the approach I had taken would not work for zig-zag lines. I will take another look. In the meantime, have you tried out the experimental setting Travel Compensate Skin Lines? This setting (and the others that become visible when the feature is enabled) attempts to stop the underextrusion that tends to occur when lines pattern is being printed and either a long travel has just occurred or the skin lines are very short (i.e. as an inside corner is approached). Might be worth a go?

PhilBaz, 2024-07-08:
@smartavionics 

Thanks for the reply Mark.

I remembered that along with your monotonic implementation that the native Cura implementation also existed. It seems to work as expected with zig-zag. So I think this request is probably DOA/moot?

I will have a look at Travel Compensate Skin Lines. Thank you!

Thanks again Mark!

PhilBaz, 2024-07-08:
@smartavionics humm...I may have spoken too soon...it appears to be doing some pretty no optimal stuff. It monotonic. but not great. Ill send a test model example before long.

smartavionics, 2024-07-09:
I've made some progress and it's now working fairly well. It's really quite tricky to implement so that the lines are printed with the minimum amount of travel. I will include it in my next release.

PhilBaz, 2024-07-09:
@smartavionics 

Ahh, fantastic! Appreciate you looking at it. Thanks again Mark.